pyrpl-fpga/pyrpl #298: pid manager gui needs more explicit names

lneuhaus, 2017-06-29:


SamuelDeleglise, 2017-08-14:
Not sure what you mean ? p->proportional i->integrator ?

lneuhaus, 2017-08-16:
not sure either. The issue arose because Clement did not intuitively understand what the settings of p or i meant. I checked that the description is rather clear (in the tooltip), but probably it is not intuitive to hold the mouse cursor still on a AttributeWidget that one doesnt understand. Updating the tutorial / documentation will certainly help. Another improvement would be plotting the transfer function for each pid widget. what do you think about the plotting option?

SamuelDeleglise, 2017-08-16:
Yes I had that in the back of my head for some time, however, I am not sure if it is not redundant with the lockbox transfer function plot that already exists.

closed 2026-09-23 by michaelcroquette