pyrpl-fpga/pyrpl #301: remaining nosetests failures in pyqt=5

lneuhaus, 2017-07-04:
Somehow the instruments seem to never stop running / writing to the config file. @SamuelDeleglise: Do you have some major changes to the acquisitioninstruments in your mind that might have caused all this?

INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname localhost
.
Estimated time per read / write operation: 2.4 ms / 2.5 ms
pyrpl.test.test_spectrum_analyzer.TestPyrpl.test_pyrpl ... ok
pyrpl.test.test_spectrum_analyzer.TestPyrpl.test_read_write_time ... ok
=======TEARING DOWN TestPyrpl===========

======================================================================
FAIL: Make sure the scope isn't continuously writing to config file,
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\Leo\Anaconda3\envs\pyrpl-env\lib\site-packages\nose\case.py", l
ine 197, in runTest
    self.test(*self.arg)
  File "c:\lneuhaus\gitHub\pyrpl\pyrpl\test\test_hardware_modules\test_scope.py"
, line 200, in test_no_write_in_config
    assert(old==new), (old, new)
AssertionError: (65, 72)

======================================================================
FAIL: pyrpl.test.test_na.TestNA.test_benchmark
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\Leo\Anaconda3\envs\pyrpl-env\lib\site-packages\nose\case.py", l
ine 197, in runTest
    self.test(*self.arg)
  File "c:\lneuhaus\gitHub\pyrpl\pyrpl\test\test_na.py", line 95, in test_benchm
ark
    "to you Red Pitaya module" % (maxduration*1000.0, duration*1000.0)
AssertionError: Na gui should take at most 16.7 ms per point, but actually needs
 73.4 ms. This won't compromise functionality but it is recommended that establi
sh a more direct ethernet connectionto you Red Pitaya module

======================================================================
FAIL: pyrpl.test.test_na.TestNA.test_na_running_states
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\Leo\Anaconda3\envs\pyrpl-env\lib\site-packages\nose\case.py", l
ine 197, in runTest
    self.test(*self.arg)
  File "c:\lneuhaus\gitHub\pyrpl\pyrpl\test\test_na.py", line 34, in test_na_run
ning_states
    assert data_changing()
AssertionError

======================================================================
FAIL: This was so hard to detect, I am making a unit test
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\Leo\Anaconda3\envs\pyrpl-env\lib\site-packages\nose\case.py", l
ine 197, in runTest
    self.test(*self.arg)
  File "c:\lneuhaus\gitHub\pyrpl\pyrpl\test\test_na.py", line 20, in test_na_sto
pped_at_startup
    assert(self.na.running_state=='stopped')
AssertionError

----------------------------------------------------------------------
XML: c:\lneuhaus\gitHub\pyrpl\xunit.xml
----------------------------------------------------------------------
Ran 352 tests in 392.112s

FAILED (failures=4)

SamuelDeleglise, 2017-07-04:
Found it!

The problem was that I changed trace_average default to 10 for the NA (to avoid unintuitive behavior after 1st trace acquisition), but the benchmark was based on a single acquisition of all trace_average.

I simply added trace_average=1 in the setup of test_benchmark to solve the problem.

Maybe we should also move all the test_na_stopped_at_startup in the setUp() function of the TestClass and make sure all tests switch off the instruments regardless of the test status because right now a failure during a test can leave a module in a running_state, thus provoking another failure in the  test_na_stopped_at_startup, which is misleading.

lneuhaus, 2017-07-04:
yeah. this suggests that we should add the option 'default' to load_state, which should establish all attributes' default values. sounds good?

SamuelDeleglise, 2017-07-04:
mmmh, why not, but in the present case, the default value is 10...

SamuelDeleglise, 2017-07-04:
It should be OK to pull now

lneuhaus, 2017-07-04:
all okay now

commit 5e9d9b6f (2017-07-04): Fixes #301
closed 2017-07-04 by lneuhaus