pyrpl-fpga/pyrpl #303: test_scope deactivated in recent release..

lneuhaus, 2017-07-05:


SamuelDeleglise, 2017-08-14:
I found the bug with _save_ctr being incremented in an unexpected and undeterministic way.

The problem was that we were counting the number of calls to _save(). However, _save can be called either by the main program execution, or by the _save_timer callback. In this case, even if nothing is saved by the continuous scope operation (what the unittest is supposed to ensure), a remaining save attempt that was pending from a previous a call (1 s earlier) can artificially increment the counter by 1.

--> The right solution (I believe) is to only count the synchronous calls to _save (because delayed saving is supposed to be an internal implementation detail of the memory tree). So I moved the actual saving activity into a function _save_no_counter() that is now called by _save(). The _save_timer is now connected to _save_no_counter such that no increment of the counter or warning is triggered by the callback.

SamuelDeleglise, 2017-08-14:
The problem with the previous solution, is that it seems to be in contradiction with the logic behind the unit_test "test_two_trees" of test_memory.py.

Anyways, I am really puzzled with this test...

SamuelDeleglise, 2017-08-14:
If we really want to test the actual count of writing to file in memory_test.py AND the number of calls to _save() from higher level modules, then I can make 2 separate counters:

_save_requested_counter
_save_tofile_counter

lneuhaus, 2017-08-16:
All this discussion concerns the above mentioned commit 66f6fa83a9ce2a63b12a2a4b0f1e2d02a919ef9f, of which I am not sure that we want to merge it without modifications.  

- I say that readability of memory.py is most important here. The above commit sacrifices readability for an ad-hoc solution to allow easier debugging. I would like to find an intermediate solution, keeping it readable and debuggable. 
- Having two or three different save functions is already painful, even more so when the docstrings are missing. Can we fix that?
- I say we reaname "_save_without_counter" to "_write_to_file" and thereby only have one "_save" function, and make sure that the logic in each function does exactly what the title suggests. then we can have counters as _save_counter and _write_to_file_counter that are incremented at the start of each function. with those changes, both of us should understand the test_two_trees function also after a few mongths.
- concerning the test_two_trees test, what puzzles you? i find the description clear, and the sequences of assertions is to ensure that MemoryTree saves to a file in exactly the way we expect it to. Of course, you having modified memory.py, the test detects a different behavior and you had to adapt it.

SamuelDeleglise, 2017-08-16:
I am ok with your suggestions, I'll implement that right away...

SamuelDeleglise, 2017-08-16:
In fact ``_save_now`` and ``_write_to_file`` are redundant.
--> In addition to what you suggested ( _save_without_counter()->_write_to_file(); _save_requested_counter->_save_counter; save_to_file_counter->_write_to_file_counter;), I removed _save_now (and replaced all occurences by ``_write_to_file``).

Unittests OK

commit 66f6fa83 (2017-08-14): Fixes #303

By introducing 2 counters
_save_requested_counter
_save_to_file_counter
closed 2017-08-16 by SamuelDeleglise