pyrpl-fpga/pyrpl #305: Easier startup procedure

lneuhaus, 2017-07-06:
We should do two things to facilitate new users getting started (especially those using the .exe distribution):

- when running from the command line (creation of PyRPL instance) fails due to missing/incorrect arguments, display a help message with the command line arguments
- pop up a start widget that allows to select an existing configuration or to set up a new one, especially proposing to enter/select a redpitaya device. The best option here would be to auto-detect the available redpitayas on the local network. It is not clear how to do this, the brute-force option being to attempt an ssh connection with user=root (and password=root, otherwise password prompt) to all ips, and then detect whether a redpitaya is responding. Detecting a redpitaya could be done by obtaining the MAC address and checking for the first four or six hex-digits. Are there better alternatives?

SamuelDeleglise, 2017-07-06:
That would be really cool. I like the MAC-address better. Or at least a prefiltering with mac-address and then try to ssh the good candidates... Of course this is a problem if ssh-ing to a redpitaya automatically disconnects other connected instances, but I believe it is not the case ?

lneuhaus, 2017-07-06:
posted a question here:
http://forum.redpitaya.com/viewtopic.php?f=8&t=2037

lneuhaus, 2017-07-06:
nmap does the job, but dont know how to integrate it in python:
https://askubuntu.com/questions/406792/list-all-mac-addresses-and-their-associated-ip-addresses-in-my-local-network-la

lneuhaus, 2017-07-06:
by the way, making an exe file with pyinstaller works like a charm if you use python3.4

lneuhaus, 2017-07-06:
Ok, there is this package, which would however add another dependency: 
https://pypi.python.org/pypi/python-nmap

Maybe we can cherry-pick the few relevant lines of code from them and work only with sockets

lneuhaus, 2017-07-06:
@SamuelDeleglise : you are interested in doing the startup window? 
I have in mind the following procedure:
0) check whether gui is True (default), if yes, the following should be windows and a message like "starting startup GUI", of no, the following should be command line input. 
1) If no config file is specified, ask what configfile name the user would like to use, including the option None for not saving a file. If a new file is created, we should offer a dialog with inputs for all the default options in pyrpl.py, including hostname, username and password. Next to the hostname, we should have a button that opens the hostname selector window. We should also show in that window how the user starts pyrpl next time with the same configuration. 
1b) The hostname selector window should have a button "Scan for Red Pitayas on local area network"" devices (or automatically do it, depending on how long it takes). Next to it, there should be a list of the options, including _FAKE_REDPITAYA_ and manual entry of ip address, and sshport=22. 
2) When selecting an option, it should be attempted to ssh to it. If the user/password from the config file/default does not work, a pop up should ask for these data. 
3) normal remaining startup procedure with better error messages in case of failed connections

SamuelDeleglise, 2017-07-06:
OK, I will try to do that over the week-end

lneuhaus, 2017-07-07:
Scan for open ssh ports (works on both my computer and the experiment - and with these settings, correctly identifies all devices (superset of redpitayas for now)):

```
import socket
port = 22
from time import sleep
def get_ip():
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    try:
        # doesn't even have to be reachable
        s.connect(('10.255.255.255', 1))
        IP = s.getsockname()[0]
    except:
        IP = '127.0.0.1'
    finally:
        s.close()
    return IP
ip=get_ip()
print ("Your own ip is: %s"%ip)
end=ip.split('.')[-1]
start = ip[:-len(end)]
ips = [start+str(i) for i in range(256)]
sockets = []
for ip in ips:
    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    s.settimeout(0.02)
    #s.setblocking(1)
    err = s.connect_ex((ip, port))
    if err == 0:
        print ("%s:%d is open"%(ip, port))
    else:
        pass
    s.close()
    #sockets.append(s)    
print ('done')
```

lneuhaus, 2017-07-07:
the hard part is to find the mac address in python without using os-dependent programs or external c-libraries.

lneuhaus, 2017-07-07:
```
import socket
port = 22
timeout=0.01
from time import sleep
def get_ip():
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    try:
        # doesn't even have to be reachable
        s.connect(('10.255.255.255', 1))
        IP = s.getsockname()[0]
    except:
        IP = '127.0.0.1'
    finally:
        s.close()
    return IP
ip=get_ip()
print ("Your own ip is: %s"%ip)
end=ip.split('.')[-1]
start = ip[:-len(end)]
ips = [start+str(i) for i in range(256)]
sockets = []
for ip in ips:
    s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    s.settimeout(timeout)
    #s.setblocking(1)
    err = s.connect_ex((ip, port))
    if err == 0:
        #print ("%s:%d is open"%(ip, port))
        s.close()
        try:
            ssh = pyrpl.sshshell.SSHshell(hostname=ip, timeout=1)
        except BaseException as e: 
            print 'Cannot log in with user=root, pw=root at', ip
        else:
            #print "root pw works"
            macs = list()
            nextgood = False
            for token in ssh.ask('ifconfig | grep HWaddr').split():
                  if nextgood and len(token.split(':')):
                      if token.startswith('00:26:32:'):
                          macs.append(token)
                          print 'RP device: ', ip, token
                  if token == 'HWaddr':
                      nextgood = True
                  else:
                      nextgood = False
            #print(macs)
            ssh.channel.close()
    else:
        s.close()
print ('done')
```

Output:

> Your own ip is: 10.214.1.43
> Cannot log in with user=root, pw=root at 10.214.1.8
> RP device:  10.214.1.9 00:26:32:F0:08:8A
> RP device:  10.214.1.15 00:26:32:F0:0C:86
> RP device:  10.214.1.20 00:26:32:F0:0E:49
> RP device:  10.214.1.21 00:26:32:F0:0E:4D
> RP device:  10.214.1.22 00:26:32:F0:0D:EC
> RP device:  10.214.1.23 00:26:32:F0:0E:4E
> RP device:  10.214.1.28 00:26:32:F0:21:47
> RP device:  10.214.1.29 00:26:32:F0:21:44
> RP device:  10.214.1.30 00:26:32:F0:19:B8
> RP device:  10.214.1.32 00:26:32:f0:17:ee
> RP device:  10.214.1.33 00:26:32:f0:20:53
> RP device:  10.214.1.34 00:26:32:f0:20:52
> RP device:  10.214.1.37 00:26:32:F0:3E:A0
> Cannot log in with user=root, pw=root at 10.214.1.254
> done

lneuhaus, 2017-07-19:
current master version is quite acceptable in terms of usability

closed 2017-07-19 by lneuhaus