pyrpl-fpga/pyrpl #307: qtpy is for the moment incompatible with pyinstruments...

ManipMembranes, 2017-07-12:
I believe this is due to the QString API version that is automatically set to version 2.0 in qtpy and which is enforced to 1.0 in pyinstruments.

This is definitely not a bug of Pyrpl, but rather of pyinstruments, but I still document it here to avoid problems in the mean time.

I am not sure how easy it will be to make pyinstruments compatible. I will probably try to make it directly pyqt3 compatible.

ManipMembranes, 2017-07-12:
Actually, yhe following is enough to have a bug:

    from PyQt4 import QtCore, QtGui
    from qtpy import QtCore, QtGui

SamuelDeleglise, 2017-07-27:
I created a branch of pyinstruments that is compatible with Python 3 and PyQt5. The branch is "pyqt5".
There is probably some bugs left with instrument drivers, but nothing terrible. The one I am most worried about is the dotnet interface with the VSA, (I am not sure if you are still using it ?). I could not find the way to have the clr module to work properly with the VSA under python 3 right now (In fact, I don't quite remember how we were supposed to do with python 2 anyways...).

lneuhaus, 2017-07-27:
clr requires version 2.0 to work properly with our code. so pip install pythonnet==2.0 used to work in python 2. does this do the job in python 3?

SamuelDeleglise, 2017-07-28:
Mmmh, unfortunately it doesn't look so simple: I get the following error. 

    Collecting pythonnet==2.0
      Using cached pythonnet-2.0.0.tar.gz
        Complete output from command python setup.py egg_info:
        Traceback (most recent call last):
          File "<string>", line 1, in <module>
          File "C:\Users\Samuel\AppData\Local\Temp\pip-build-j2mc3a65\pythonnet\setup.py", line 79, in <module>
            _xbuild = "\"%s\"" % _find_msbuild_tool("msbuild.exe")
          File "C:\Users\Samuel\AppData\Local\Temp\pip-build-j2mc3a65\pythonnet\setup.py", line 27, in _find_msbuild_tool
            import _winreg
        ImportError: No module named '_winreg'

        ----------------------------------------
    Command "python setup.py egg_info" failed with error code 1 in 
    C:\Users\Samuel\AppData\Local\Temp\pip-build-j2mc3a65\pythonnet\

Looking at the pypi page for pythonnet, the support for python 3 only seems to begin at version 2.1 (pip install pythonnet==2.1 works like a charm, but crashes with the VSA...).

lneuhaus, 2017-07-31:
So, this makes a full port of pyinstruments to python 3 depend on our ability to get the VSA driver running in python 3 (maybe there is another package that allows using the dotnet interface in python 3)?

However, a totally different issue, which prevents using the old version of  pyinstruments with the new pyrpl version, is the above-stated error 

```
from PyQt4 import QtCore, QtGui
from qtpy import QtCore, QtGui
```

Maybe we can just fix that one?

lneuhaus, 2017-07-31:
I tried:
```
conda create -y -n py27qt4 python=2.7 numpy scipy paramiko pandas nose pip pyqt=4 qtpy pyqtgraph pyyaml
activate py27qt4
python
# in the python console:
from qtpy import QtCore, QtGui
from PyQt4 import QtCore, QtGui
# no error appears. However, if I swap the order of the imports, an error message related to QString is there. But this can visibly be avoided without big efforts.
```

Then, I also tried 
```
conda install django= MySQL-python h5py pytables
pip install guidata
pip install south
pip install guiqwt==3.0.2
pip install matplotlib
pip install django_model_utils
pip install tablib
pip install django_import_export
pip install django==1.6.5

python
import pyrpl
from pyinstruments import CurveDB
```

However, I get errors when I try to create a Curve. So instead of using pyinstruments master, I have been using pyinstruments branch pyqt5. After small bugfixes in python 2.7, I can do the following:

```
python
import pyrpl
from pyinstruments import CurveDB
p=pyrpl.Pyrpl()
c = CurveDB.create([],[])
```

I think this is all we need to continue using the most recent pyrpl with pyinstruments in python 2.7.

lneuhaus, 2017-07-31:
I am in favour or pushing pyinstruments.pyqt5 to pyinstruments.master

denfromufa, 2017-08-01:
@lneuhaus why do you require pythonnet 2.0, can you point to any specific regression?

lneuhaus, 2017-08-01:
@denfromufa Thanks for joining the discussion. Just for info, this issue is mostly related to another Python package "pyinstruments" which is not public at the moment. 

The dependency on pythonnet 2.0 and incompatibility with higher versions was found by quick trial and error, and is purely related to trying to make the dotnet driver for the [Keysight VSA Software (version 19.5)](http://www.keysight.com/main/software.jspx?cc=DE&lc=ger&nid=-32806.0.02&id=2013472&pageMode=PV) work. 

If you like, I will post an issue with the actual error message in https://github.com/pythonnet/pythonnet/issues whenever the problem comes up again.

lneuhaus, 2017-08-01:
@SamuelDeleglise do you have the VSA software error message with clr version >= 2.1 at hand?

SamuelDeleglise, 2017-08-07:
I will try to post the error message later today, however, I can maybe clarify what I did with pyinstruments.

In fact the branch pyqt5 is at tje same tie supposed to work with pyqt5 and python3. In fact, I didn t test at all the compatibility witj python 2.7, but as you said, it should be straight forward to make it compatible with both python2 and python 3. 
The good news is that the python 3 version seems to work now with the lattest version of most libraries (django, qwt...). I have already made a python 3 environment on our experiebt computer and it seems to be reasonably bug free now. The only remaining issue is the vsa software, but we are not using it right now

lneuhaus, 2017-08-07:
Hi Samuel, 
I think i pushed the two small corrections for python 2.7 to pyinstruments, so it should equally well work in that version. However it today seems that pyinstruments is disabled:
[image]
Did you do that or is it an error?

SamuelDeleglise, 2017-08-07:
Shit, I believe it s because my educational offer for github has expired. I asked for a renewal today... Another option would be to make the repository public

lneuhaus, 2017-08-07:
Whatever optioin you prefer. I personally do not think that there is a great problem if it is made public, and a small amount of pyrpl users might benefit from pyinstruments. Also, the installation issues are somewhat manageable now with Anaconda, in particular given the relatively small amount of potential users (the fear of using another group's - possibly very buggy - software is pretty high imho, unless it is really well documented and established).

SamuelDeleglise, 2017-08-11:
OK, I checked again, with clr>=2.1 there is not even an error message since the whole python interpreter crashes. You just get the windows pop-up "Python a cessé de fonctionner"

In the kernel console, I get the following:

    Exception non gérée : System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
       à Python.Runtime.ClassDerivedObject.ToPython(IPythonDerivedType obj)
       à Python.Runtime.Converter.ToPython(Object value, Type type)
       à Python.Runtime.MethodObject.Invoke(IntPtr target, IntPtr args, IntPtr kw, MethodBase info)
       à Python.Runtime.MethodBinding.tp_call(IntPtr ob, IntPtr args, IntPtr kw)

SamuelDeleglise, 2017-08-11:
I guess github reopened my unlimited private repository plan in the mean time. Do you have access to pyinstruments again ?

lneuhaus, 2017-08-14:
seems this is okay now. can we close this issue?

closed 2017-08-14 by SamuelDeleglise