pyrpl-fpga/pyrpl #309: iir user experience

lneuhaus, 2017-08-09:
[] currently, after each select, the entire filter is recomputed about 7 times (too many calls to _setup), which slows down the code.
[] need to allow for tiny changes with short keyboard keystrokes
[] need to plot ratios etc, possibly legend

SamuelDeleglise, 2018-12-18:
I played a bit with the repaired iir interface and it looks very nice and cool with the button "measure transfer function". However, it took me some time to understand how to obtain a transfer function that looks like the designed curve. In particular, I didn't realize at the beginning that the gain parameter was the "magic button". 
If I understand correctly, below a certain threshold, you hit rounding errors in the filter calculations, and above a certain threshold, you reach an overflow in there ? Why are you probing with such a low na amplitude ? Is it that you want the filter to be efficient for small fluctuations and don't really care about large excursions as long as you are close to the locking point ?
I believe it would be helpful for beginner users like me to start with a reasonable default gain (such as 10^3) because the first transfer function I got were so crazy with a gain of 1 that I almost thought the module was bugged.

lneuhaus, 2018-12-19:
There are actually two magic buttons: gain and loops (loops should be swapped for sampling_rate which is proportional to its inverse). 
You are correct that gain should be as large as possible as long as the raw filter coefficients dont saturate (i think the maximum is 4 or 8 at the moment, and we could in principle change that) to provide the smallest rounding errors. We could actually add a global gain to the IIR such that it will always be implemented with the best possible coefficient resolution to make it more user-friendly. 
In principle feel free to change the default values such as gain or na settings. 

However, to explain the current values: A typical IIR filter will have a maximum peak height about 40 dB above its DC-response (gain=DC-response). Thus with a gain of 1, the NA will saturate the filter for amplitudes above 10 mV. 

If you had to set a gain of 10**3 to get something reasonable, I suspect a big. Can you attach a screenshot of the IIR interface with your "naive" settings that did not work? 

By the way, have you seen that the NA settings can be customized by modifying the na state "iir_measurement"?