pyrpl-fpga/pyrpl #322: Change widget background color

woshibala, 2017-10-16:
Hi there, 
I am working on adding a new module and widget to show the Allen Dev. I was wondering how can I use your spectrum module and let it plot in a fixed interval.

Also, The plot background is black, and I was trying to change it into white. But I can find where I can make this change, or there is no background for plot and I need to add a background to it? 

Thanks,

lneuhaus, 2017-10-17:
Hi, 
welcome to the developers! If you are thinking about sharing your modifications with us or others in the future, I highly recommend you to fork the repository before modifying it. This will make discussing your code easier in the future. 

Also, the current documentation is definitely not at a state that would allow you to reasonable develop code based on that. We will be fixing that soon (hopefully), but if you are in a hurry, dont hesitate to ask a lot of questions here or you might have to read a lot of unuseful code for your purpose. 
 
Now your first question:

If you simply want the spectrum analyzer to plot a new curve every so and so seconds, the easiest way is to periodically call (`sa` is the specan module here, set `sa.trace_averages=1` before if you do not want to average several traces):
```
sa.single()  # blocking until spectrum acquisition is finished
```
or
```
sa.single_async()  # running in the background, returns a 'future'
```
We have a module called `Loop` (see the file loop.py) for periodic calls which you might want to use for this, otherwise QTimers are fine, too. 

If you could be a little more specific, maybe with some of your code (if there is already any), I can probably extend this answer on the topic that needs more explanation.

Then, your second question about the background:

We use pyqtgraph for all plots (with default configuration), and the relevant commands seem to be documented here:
http://www.pyqtgraph.org/documentation/style.html#default-background-and-foreground-colors

Copy-paste:
```import pyqtgraph as pg

## Switch to using white background and black foreground
pg.setConfigOption('background', 'w')
pg.setConfigOption('foreground', 'k')
```
(Note that this must be set before creating any widgets)

What exact code snippet do you use to plot? With that information, I could suggest a nice place to insert this into the code if you want me to..

woshibala, 2017-10-17:
Hi,

Thanks for your reply. I tried to change the background according to your answer, and it works! I am going to look into the loop.py and QTimers.

I am also working on plot data on polar coordinates, and it rather different from any pyrpl widget. Do you have any suggestion how can I implement polar coordinates based on your work?

Thanks again! Have a good day.

lneuhaus, 2017-10-17:
Cool. loop.py is essentially just a wrapper around QTimers, so reading the code might be a good example to see how QTimers can be used, and then to decide whether to use the wrapper or a custom implementation. 

For polar coordinates I have not much to say. A quick search revealed this proposition which seems to work with pyqtgraph:
https://groups.google.com/forum/#!topic/pyqtgraph/9Vv1kJdxE6U 

Otherwise, if you do not plot too frequently, the somewhat slower plots produced with matplotlib should do the job as well.

closed 2017-11-14 by lneuhaus