pyrpl-fpga/pyrpl #323: scope module freezed when run in Mac

greenhk, 2017-10-21:
Dear Dr. Ineuhaus,
Thanks a lot for this wonderful work. It runs well in Windows, but we found that when it runs in Mac, the scope module will be freezed after launching acquisition. If possible, could you please prepare me an executable version in Mac [macOS Sierra 10.12.6]. Thanks.

lneuhaus, 2017-10-21:
Hi Greenk,

greenhk, 2017-10-21:
Hi,
Thanks for your prompt response. Here are your required information:
- I'm using Python 2.7.13, qt 4.8.7_2
- I install pyrpl by "pip install pyrpl or pip2 install pyrpl"
- I start by "python -m pyrpl example_filename"
- see the yml file below

Thank you so much.

----------------
pyrpl:
  background_color: ''
  dock_positions: "\0\0\0\xFF\0\0\0\0\xFD\0\0\0\x01\0\0\0\x02\0\0\x03 \0\0\x02D\xFC\
    \x01\0\0\0\x01\xFC\0\0\0\0\0\0\x03 \0\0\0h\0\xFF\xFF\xFF\xFA\0\0\0\x03\x01\0\0\
    \0\v\xFB\0\0\0.\0a\0s\0g\0s\0 \0(\0e\0x\0a\0m\0p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0\
    m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0\x1A\0\xFF\xFF\xFF\xFB\0\0\0,\0i\0q\0s\0\
    \ \0(\0e\0x\0a\0m\0p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\
    \xFF\0\0\0\x1A\0\xFF\xFF\xFF\xFB\0\0\0.\0p\0i\0d\0s\0 \0(\0e\0x\0a\0m\0p\0l\0\
    e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0\x1A\0\xFF\xFF\xFF\
    \xFB\0\0\02\0s\0c\0o\0p\0e\0s\0 \0(\0e\0x\0a\0m\0p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0\
    m\0e\0)\x01\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0h\0\xFF\xFF\xFF\xFB\0\0\0.\0i\0i\0r\0\
    s\0 \0(\0e\0x\0a\0m\0p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\
    \xFF\0\0\0\x1A\0\xFF\xFF\xFF\xFB\0\0\00\0t\0r\0i\0g\0s\0 \0(\0e\0x\0a\0m\0p\0\
    l\0e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0\x1A\0\xFF\xFF\
    \xFF\xFB\0\0\0D\0n\0e\0t\0w\0o\0r\0k\0a\0n\0a\0l\0y\0z\0e\0r\0 \0(\0e\0x\0a\0\
    m\0p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0\x1A\0\
    \xFF\xFF\xFF\xFB\0\0\0F\0s\0p\0e\0c\0t\0r\0u\0m\0a\0n\0a\0l\0y\0z\0e\0r\0 \0(\0\
    e\0x\0a\0m\0p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\
    \0\x1A\0\xFF\xFF\xFF\xFB\0\0\0<\0c\0u\0r\0v\0e\0v\0i\0e\0w\0e\0r\0 \0(\0e\0x\0\
    a\0m\0p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0\x1A\
    \0\xFF\xFF\xFF\xFB\0\0\0<\0p\0y\0r\0p\0l\0c\0o\0n\0f\0i\0g\0 \0(\0e\0x\0a\0m\0\
    p\0l\0e\0_\0f\0i\0l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0\x1A\0\xFF\
    \xFF\xFF\xFB\0\0\04\0l\0o\0c\0k\0b\0o\0x\0 \0(\0e\0x\0a\0m\0p\0l\0e\0_\0f\0i\0\
    l\0e\0n\0a\0m\0e\0)\0\0\0\0\0\xFF\xFF\xFF\xFF\0\0\0\x1A\0\xFF\xFF\xFF\0\0\x03\
    \ \0\0\0\0\0\0\0\x04\0\0\0\x04\0\0\0\b\0\0\0\b\xFC\0\0\0\0"
  gui: true
  loglevel: info
  modules:
  - NetworkAnalyzer
  - SpectrumAnalyzer
  - CurveViewer
  - PyrplConfig
  - Lockbox
  name: example_filename
  window_position:
  - 334
  - 23
  - 800
  - 600
redpitaya:
  autostart: true
  delay: 0.05
  frequency_correction: 1.0
  hostname: _FAKE_
  leds_off: true
  monitor_server_name: monitor_server
  password: root
  port: 2222
  reloadfpga: true
  reloadserver: false
  serverbinfilename: fpga.bin
  serverdirname: //opt//pyrpl//
  silence_env: false
  sshport: 22
  timeout: 1
  user: root
scope:
  curve_name: scope curve
networkanalyzer:
  curve_name: networkanalyzer curve
spectrumanalyzer:
  curve_name: spectrumanalyzer curve
lockbox:
  sequence:
  - outputs: {}
--------------------

lneuhaus, 2017-10-21:
Cool, thanks a lot. Looks like it is probably a python 2.7 bug, since I could reproduce it with the following:

```
conda create -y -n py27debug python=2.7.13 numpy scipy paramiko pandas nose pip pyqt qtpy pyqtgraph pyyaml
activate py27debug
pip install pyrpl
# i had to do this twice to make it work (possibly a problem with my local PYTHONPATH setting and older pyrpl installations)
pip install pyrpl
# copy-pasted your yml to a file named py27debug.yml
python -m pyrpl py27debug
```
Here, the scope did not freeze on my windows machine. Then i downgraded my PyQt version from 5.6 to 4.11 with
```
conda install pyqt=4
```
Still working (with qt version 4.11). Then I tried the lowest pyqt installation I could find with conda
```
conda install pyqt=4.10
```
Still working. Even though this was only on windows so far, its getting too difficult to build the exact same python installation with all the same library versions, so I'm tempted to giving up on trying to reproduce the error and blame it on an outdated python installation. 

Two consequences:

- Either update the python installation on the mac you are running pyrpl on, at least the pyqt version for a start. This has good chances to work, since I have successfully run pyrpl on a mac before - with anaconda and
```
conda create -y -n pyrpl-env numpy scipy paramiko pandas nose pip pyqt qtpy pyqtgraph pyyaml
activate pyrpl-env
```
- If you dont want or can, wait until tomorrow and I will compile a binary for mac (hopefully the right OS version). 

For the next version upgrade, we will add the mac binary to the auto-deploy step (possible in travis, see https://docs.travis-ci.com/user/multi-os/).

greenhk, 2017-10-22:
Thanks a lot for these help. I think the freezing may be linked to the pyqtgraph, since everything is smooth before clicking the "Run continuous". The same problem also occurs in the spetrumanalyzer module. I just tried to install everything in a newly built python virtual environment, but the issue is still there.  I also notice that after a long time (about several minutes), the scope will eventually show one trace of data, and then freeze again. It seems that it's waiting something.

greenhk, 2017-10-22:
Cool, finally the problem is solved by a new installation of python from Anaconda (before the python 2.7 is installed from Brew), although I don't know the exact reason. Now I can have some fun to play with it. Thank you again for your hints :)

closed 2017-10-22 by greenhk

lneuhaus, 2017-10-22:
Nice. Anyways, the MacOS binaries are on their way...