pyrpl-fpga/pyrpl #324: MacOSX binary not working

lneuhaus, 2017-10-22:
Error message:

This application failed to start because it could not find or load the Qt platform plugin "cocoa"
in "".

Reinstalling the application may fix this problem.
Abort trap: 6

lneuhaus, 2017-10-22:
The answer is here: https://stackoverflow.com/questions/25188153/this-application-failed-to-start-because-it-could-not-find-or-load-the-qt-platfo

lneuhaus, 2017-10-26:
The real issue is with pyinstaller and is a known issue: 
https://github.com/pyinstaller/pyinstaller/issues/2857

They kept on trying to fix it for a while but it seems like the solution is only going forth and back. Let's wait for the next release and ad-hoc fix it then if necessary.

lneuhaus, 2017-10-26:
417e8e77af6f703af9c15f181b075c65b6b2637a

lneuhaus, 2017-11-14:
Current error message: 
```
Traceback (most recent call last):
  File "PyInstaller/loader/rthooks/pyi_rth_pkgres.py", line 11, in <module>
  File "/Users/travis/build/lneuhaus/pyinstaller/PyInstaller/loader/pyimod03_importers.py", line 631, in exec_module
  File "site-packages/pkg_resources/__init__.py", line 1040, in <module>
  File "site-packages/pkg_resources/__init__.py", line 1043, in Environment
  File "site-packages/pkg_resources/__init__.py", line 272, in get_supported_platform
  File "site-packages/pkg_resources/__init__.py", line 478, in get_build_platform
  File "sysconfig.py", line 688, in get_platform
  File "sysconfig.py", line 549, in get_config_vars
  File "sysconfig.py", line 420, in _init_posix
ImportError: No module named '_sysconfigdata_m_darwin_'
[8831] Failed to execute script pyi_rth_pkgres
```

fbbarnes, 2022-01-18:
Hi, I've been coming across this error using the pyrpl-osx executable on sourceforge. I've also tried making an executable using the instructions [here](https://github.com/lneuhaus/pyrpl/blob/ae31209a7f92d0fea3b269ea39a017b650f36a93/docs/source/developer_guide/dist/executable.rst), but it seems that these instructions are out of date. I had difficulty understanding how to follow the answer on Stackoverflow so if you could help with a solution that would be great. Cheers!

closed 2026-09-23 by michaelcroquette