pyrpl-fpga/pyrpl #332: Analog PID setpoint

plauria, 2018-02-20:
Hi guys,

For arbitrary laser intensity ramps, we would love to be able to make the PID setpoint equal to an analog input -- from either an RF input or an XADC input. RF input would be ideal as it is fastest. 

This doesn't seem possible with the current FPGA firmware, although I might try my hand at recompiling it with this addition, as it seems like it could be very easy to implement.  (?) Alas I have little verilog experience. . . 

Thanks!

lneuhaus, 2018-02-20:
Hi

The easiest way to get what you want is to use two PID modules with the
same output_direct. If you set the gains of the second PID module (p and i)
to equal values with opposite sign as the first one, the output will be the
same as the difference signal fed to a single PID. There will be no
bandwidth limit apart from the integrator.

A cleaner solution would be a difference block before yhe PID, but i
currently have no time to develop that.

kai-niklas, 2018-03-15:
Hi,
did this suggestion work for anyone? 
For me this worked when just using P, but I had trouble with railing integrators.

Thanks!

plauria, 2018-03-16:
It didnt. I had the same issue with the ingegrator, and I was never happy with the proportional gains. 

 It's not really a proper solution because what you want is for the addition to occur *before* the PID algorithm is applied. 
What I ended up doing was making an analog summing amplifier, summing my error signal with an ASG which then goes into the redpitaya PID.

lneuhaus, 2018-07-01:
Since this is the number 1 feature request, I am starting a wishlist here:

Wishes from JP: 

- [ ] One differential PID per redpitaya is enough: "for our application, there is no point is trying to control several laser powers with a single RP since there are only two inputs (photodiode in and analog voltage control). A single PID controller using the difference as error signal would do the job for 95% of the cases. "
- [ ] In differential mode, use two PID's together so that two outputs with different gains are possible: "The ultimate feature would be to be able to set two different sets of gains for the two outputs, based on the same error signal (for exemple, to have a fast feedback on an AOM RF amplitude, and a slow control over a rotating lambda plate)"
- [ ] Keep the constant setpoint: "Offset compensation on the error signal is also important, even though we can cheat on that by offsetting the analog control signal. 
- [ ] Gains of +1 and -1 in differential mode (before PID) is enough: "Having a different gain on the two inputs is not so important since photodiodes usually feature an tunable gain. "

Any other suggestions? Otherwise we can simply make a differential PID module out of pid0 and pid1 which can be switched from individual (as before, as two separate PIDs) and differential mode.

lneuhaus, 2018-12-11:
on branch develop-0.9.3, above features (checkboxes) are implemented as-is. In the future, we might even add custom gains instead of +-1 for the difference signal. One question that is unclear is where the input_filter should act: before or after the difference. 

Before: advantage is that both inputs can be filtered with different cutoffs. 
After: advantage is that the two gains can be equipped with different cutoffs.

Anyone has a preference here?

e3nder, 2019-04-27:
Since there wasn't a lot of documentation surrounding this new feature we had a little bit of trouble understanding how to use it.

Generally what we were trying to do was have a setpoint ramp for a PI.  Setting up the differential mode seemed to work, feeding the detector into one PID module and an ASG into another. However we had a lot of trouble with setting the magnitude of the gains correctly.  It always felt like the ASG was significantly larger in magnitude compared to the 'in1'. Could you add a bit more into the docstring to explain the differential mode and what it is calculating?  Is it really just calculating in1 - in2 and sending that to the filters of pid0?

closed 2026-09-23 by michaelcroquette