pyrpl-fpga/pyrpl #333: Cannot type in anything, linux pyrpl binary

plauria, 2018-02-26:
What I see on fedora 27: 

$ ./pyrpl-linux 
xkbcommon: ERROR: failed to add default include path /home/travis/miniconda/envs/test-environment/lib
Qt: Failed to create XKB context!
Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
0 ./pyrpl-linux
Calling Pyrpl(**{})
INFO:pyrpl:Please select or create a configuration file in the file selector window!
Fontconfig error: Cannot load default config file

And then I cannot type in a hostname, or anything in any dialog boxes, nothing happens.

lneuhaus, 2018-02-26:
I am unfortunately unable to reproduce you error. But searching for it in google gets me to stackoverflow, where this bug seems to be well-known:
 https://stackoverflow.com/questions/26974644/no-keyboard-input-in-qt-creator-after-update-to-qt5

Some are able to solve it with 

```export QT_XKB_CONFIG_ROOT=/usr/share/X11/xkb
qtcreator```

If this doesnt work and you just want to start pyrpl (without needing the keyboard later during program execution), I suggest you can try to pass all the stuff as command-line arguments:
```pyrpl-linux config=myconfigfilename_without_yml hostname=192.168.1.100```
where, of course, you should put your symbols in. 

Running pyrpl from python (not the binaries) should also not have this issue. 

I will leave this issue open for some time in case someone stumbles upon the problem again, and in that case try to

plauria, 2018-02-26:
Hi

See

https://lists.debian.org/debian-backports/2014/10/msg00061.html

Setting export QT_XKB_CONFIG_ROOT=/usr/share/X11/xkb fixes it for me.

plauria, 2018-02-26:
By the way for some reason I have to run pyrpl binary as sudo or it will fail to connect with lots of errors like this: 
ERROR:paramiko.transport:Exception: key cannot be used for signing
ERROR:paramiko.transport:Traceback (most recent call last):
ERROR:paramiko.transport:  File "site-packages/paramiko/transport.py", line 1791, in run
ERROR:paramiko.transport:  File "site-packages/paramiko/auth_handler.py", line 241, in _parse_service_accept
ERROR:paramiko.transport:  File "site-packages/paramiko/agent.py", line 400, in sign_ssh_data
ERROR:paramiko.transport:paramiko.ssh_exception.SSHException: key cannot be used for signing
ERROR:paramiko.transport:
...
Could not connect to the Red Pitaya device with the following parameters: 

	hostname: 172.16.74.45
	ssh port: 22
	username: root
	password: ****
Please confirm that the device is reachable by typing its hostname/ip address into a web browser and checking that a page is displayed. 

It works fine if I use sudo.

lneuhaus, 2018-02-26:
this seems to be related to https://github.com/paramiko/paramiko/issues/391. i will test the solution therein (`ssh.connect("remotehost", username="myuser",password="mypassword", look_for_keys=False)`) and see if it avoids that issue.

closed 2026-09-23 by michaelcroquette