pyrpl-fpga/pyrpl #334: Qt4 and Qt5 side-by-side installation breaks pyrpl?

plauria, 2018-02-26:
Hi, sorry for the bugspam!  ;)  I don't mean for this to be my personal support forum but I suspect this may be an actual issue with some machines. 

I tried installing pyrpl from pip, and cloned the git respository. I get import errors for pyrpl:

```

import pyrpl
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-1-f256f7007682> in <module>()
----> 1 import pyrpl

/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/__init__.py in <module>()
     31 
     32 # get QApplication instance
---> 33 from qtpy import QtCore, QtWidgets
     34 APP = QtWidgets.QApplication.instance()
     35 if APP is None:

/usr/lib64/python2.7/site-packages/qtpy/__init__.py in <module>()
    123         import sip
    124         try:
--> 125             sip.setapi('QString', 2)
    126             sip.setapi('QVariant', 2)
    127             sip.setapi('QDate', 2)

ValueError: API 'QString' has already been set to version 1

```

That's from ipython. From regular python2:
```

[paul@strontium ~]$ python test.py 
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import pyrpl 
  File "/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/__init__.py", line 66, in <module>
    from .redpitaya import RedPitaya
  File "/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/redpitaya.py", line 19, in <module>
    from . import redpitaya_client
  File "/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/redpitaya_client.py", line 28, in <module>
    from .hardware_modules.dsp import dsp_addr_base, DSP_INPUTS
  File "/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/hardware_modules/__init__.py", line 23, in <module>
    from .dsp import DspModule, DSP_INPUTS, all_inputs, all_output_directs, \
  File "/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/hardware_modules/dsp.py", line 2, in <module>
    from ..attributes import BoolRegister, SelectProperty, SelectProperty, SelectRegister
  File "/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/attributes.py", line 20, in <module>
    from .widgets.attribute_widgets import BoolAttributeWidget, \
  File "/usr/lib/python2.7/site-packages/pyrpl-0.9.3.6-py2.7.egg/pyrpl/widgets/attribute_widgets.py", line 10, in <module>
    import pyqtgraph as pg
  File "/usr/lib/python2.7/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui
  File "/usr/lib/python2.7/site-packages/pyqtgraph/Qt.py", line 104, in <module>
    from PyQt4 import QtGui, QtCore, uic
RuntimeError: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class

```

I need PyQt4 for some older apps so I can't uninstall that... it seems pyrpl is getting confused because of this. Any ideas?

lneuhaus, 2018-02-26:
No prob, I am happy if we discuss all issues rather sooner than later.

its been a while i had these issues. if i remember correctly, this bug can be worked around by either loading pyqt4 or pyqt5 before attempting to load pyrpl. so maybe playing with these combinations will get you somewhere:

```
import PyQt4
import pyrpl
```

```
import PyQt5
import pyrpl
```

```
from PyQt4 import QtCore
import pyrpl
```

or something like that. At least i believe the error message should change. 

Another thing to try in ipython before loading pyrpl: 
```
%gui qt
import pyrpl
```

```
%gui qt4
import pyrpl
```

```
%gui qt5
import pyrpl
```

good luck

plauria, 2018-02-26:
Hi
Unfortunately none of those work for me. They all give the same error regarding API version, except for the last one; apparently pyqtgraph requires qt4. 

Exception: PyQtGraph requires either PyQt4 or PySide; neither package could be imported.

plauria, 2018-02-27:
Is there any easy way I can disable the loading of all the gui elements? I really just need command line control of the RP. Thanks

plauria, 2018-02-27:
I figured it out. Since pyqtgraph requires qt4, that has to be used. That's fine.

So pyrpl/__init__.py line 28 should be (explicitly)
`IPYTHON.magic("gui qt4")`

In addition, to force the loading of qt4 and NOT qt5 on line 34, first do:
```import os
os.environ['QT_API'] = 'pyqt'
```
This forces qtpy to load qt4, otherwise it will by default load qt5.

lneuhaus, 2018-02-27:
Cool thanks! only loading qt4 will not work since many people only have qt5. But maybe we can find another workaround, i will have to play around a little.

plauria, 2018-02-27:
It's kind of weird, the above trick only works if I do the following in ipython BEFORE calling import pyrpl:
```

import os
os.environ['QT_API'] = 'pyqt'
from qtpy import QtCore, QtWidgets

```
It apparently does no good chaning the __init__ to have the above lines, which makes no sense to me.

michaelcroquette, 2026-10-01:
Closing as part of a triage of long-inactive issues: Qt4 is no longer supported. PyRPL now uses qtpy with PyQt5, PyQt6, PySide2 or PySide6. If this still affects you with the current release (0.9.9.1 / main), please comment here and we will reopen.

closed 2026-10-01 by michaelcroquette