pyrpl-fpga/pyrpl #335: asg to generate custom waveforms?

plauria, 2018-02-28:
Hi, maybe this is a documentation issue only and not a request, can you please clarify? 

It doesn't seem like it's currently possible for the "arbitrary" signal generator to generate actually _arbitrary_ signals. I tried setting asg.data = [some custom ndarray], but the actual output only wants to give me sine waves. 

The waveform list is limited to a few standard waveforms (cosines and ramps etc.) Shouldn't there be a "custom" waveform? 

Is there some way to easily do this?  I remember the old redpitaya apps had this feature where you could upload a text document. They removed it in later versions of their gui, which is one of the reasons I don't use theirs anymore. 

This is probably straightforward, but I didn't see how to do this from the API docs. Thanks!

plauria, 2018-02-28:
Hi, sorry, I feel kind of silly. It does work as I expect. But maybe it should be noted in the docs somewhere. For others looking;
Arbitrary signals are on: 
asg.data = np.ndarray(your_data_array) 

asg.waveform doesn't return correct results if you call asg.data  directly. That's why I was getting confused.

lneuhaus, 2018-02-28:
I suggest we could modidy the API as follows: asg.waveform=np.array... sets
a custom waveform to asg.data and returns "custom" for print(asg.waveform).

This would keep asg.data more low-level to avoid above confusion. We coild
even rename data to _data and have a public, readonly property data to read
the data.

What do you think?

plauria, 2018-02-28:
I agree, I like asg.waveform = np.ndarray(...); it's good to keep data low level. 

By the way, it seems like line 55 in asg.py will never be executed - a ValueError on line 18 will happen before that.

plauria, 2018-03-01:
Also passing a non-numpy array should raise an exception. I tried passing a list by mistake and get obscure errors:`
ERROR:pyrpl.redpitaya_client:Error occured in reading attempt 0. Reconnecting at addr 0x40210000 to _writes value [1 1 1 ..., 0 0 0] by client 26`

plauria, 2018-03-01:
Another issue: changing asg.trigger_source from immediate to off, then back again, changes asg.data to the waveform described by asg.waveform.

e3nder, 2019-04-25:
I don't want to pile on, but in general setting up custom waveforms is pretty unintuitive.
`asg.waveform = 'custom2'`

`ERROR:pyrpl.attributes:Value 'custom2' is not an option for SelectAttribute waveform of module asg0 with options OrderedDict([('sin', 'sin'), ('cos', 'cos'), ('ramp', 'ramp'), ('halframp', 'halframp'), ('square', 'square'), ('dc', 'dc'), ('noise', 'noise')])`

I've temporarily fixed this by having a "custom" option specified in `waveforms` when it is building the asg.  Is there a way to add other waveforms without having to edit the code?
`

lneuhaus, 2019-04-27:
you mean by using the API, such as a command `asg.add_waveform(data)` or the like?

lneuhaus, 2019-04-27:
the only currently working solution is to directly write to asg.data, which however 
- leaves the waveform name as it was before
- only works if waveform='noise' is not selected when the custom data is set

```
import numpy as np
x = np.linspace(0, 2*np.pi, asg.data_length, endpoint=False)
y = np.sin(x)
asg.data = y
```

e3nder, 2019-04-27:
I think using an add_waveform is better.

The main issue is that leaving the waveform with an incorrect name, it's possible to accidentally have it recreate the incorrect waveform. I don't have it in front of me right now but I remember when testing that just writing the data, but leaving the `'sin'` could somehow accidentally produce a sine wave. Maybe it was when changing triggers?

e3nder, 2019-04-27:
An add_waveform that alleviates this issue, so it would need to assign a name as well.