pyrpl-fpga/pyrpl #336: output "lock on" menu not appearing windows binary

tymomboe, 2018-03-01:
In any of the lock box modules (GUI, Windows binary), the output1 or output2 sections of any locking stage are blank.  They should contain a 3 state checkbox labelled "lock on" and another checkbox with "reset" followed by a text box for "offset".

I have tried: 
- starting a new config file
- uninstalling any python versions on the computer in case there is some weird conflict.
- removing all folder and files that contain "pyrpl" in them and relaunching pyrpl.  
- trying a different computer.  2 out of the 3 computers have this bug, but the third (which is a brand new computer) does not, suggesting something is up.  However, the command lines for boot up of pyrpl are identical, no funny messages.

tymomboe, 2018-03-01:
Hmmm I have tried the same executable on a fourth computer, and it functions as it should.  So far the menu has failed to appear on 2 laptops, and appeared normally on two desktops.  I'm beginning to think its something to do with the display settings on the laptops.  Could it be weird scaling issues in Windows 10?

tymomboe, 2018-03-01:
Fixed!  Turning display scaling off on the laptop fixes the issue.  I just have to turn the resolution down so all my icons aren't super tiny.

Microsoft's terrible display scaling strikes again.

lneuhaus, 2018-03-02:
Hi, I have seen the same effect before, but wasnt bound to the computer with the "bad" display (I have seen this once on a windows 10 and once on a Mac, I think). It would definitely be good to find a fix for this that works on any computer. Possibilities to fix it

1) One can try to play with the environment variable `QT_AUTO_SCREEN_SCALE_FACTOR=0` before loading QT (there is also a qt attribute somewhere), this would probaby recover the same display style as in windows 7, see https://stackoverflow.com/questions/35816944/improper-scaling-of-qt-desktop-application-in-windows-10

2) Starting from 1) with the opposite setting `QT_AUTO_SCREEN_SCALE_FACTOR=0`, we should fix the QT settings that, at least, all items of the GUI are equally nice/bad. But maybe we are already happy with 1)

lneuhaus, 2018-03-02:
On my windows 7, the variable has no influence on the display properties. So I will need to find another computer first

michaelcroquette, 2026-10-01:
Closing as part of a triage of long-inactive issues: this concerns the 2018 Windows binary, which has since been replaced. If this still affects you with the current release (0.9.9.1 / main), please comment here and we will reopen.

closed 2026-10-01 by michaelcroquette