pyrpl-fpga/pyrpl #338: Scope trigger not working after updating conda --all

quantumavik, 2018-04-19:
Hi!
The scopes module gets frozen in the GUI after updating conda and all related packages. Windows 10, Anaconda 4.5.1, Python 3.6.5.
I also get the following warning:
```
C:\ProgramData\Anaconda3\lib\site-packages\h5py\__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
```
The rolling untriggered mode still works for decimations >=1024 (>=0.13s), but the normal trigger does not. It freezes the moment I change the decimation to 512 or less, or reduce the acquisition window time to less than 0.06 s. 
Any idea what might be the cause of the problem and how to fix it?

Thanks!
Avik
PS. pyrpl is great and I've been using it regularly. It's a great interface which allows low and high-level control... Thanks :)

quantumavik, 2018-04-19:
Oh, I forgot to add that I attempted many options for the trigger source to get it working - 'immediately', 'off', 'asg0' (where asg0 was set to output a 0.2 V amplitude ramp at 100 Hz, which is displayed as expected in the rolling mode).
It is not just a problem with the GUI but the system in general, since the issue is not fixed by running the commands from the ipython interpreter either.

lneuhaus, 2018-07-01:
Hi, it is hard for me to answer you question. The easiest way to find the bug would be if I could reproduce your problem. Could you let me know how exactly you create your conda environment, i.e. send me the content of spec-file.txt after executing

```conda list --explicit > spec-file.txt```

in your conda environment, or alternatively, tell me how exactly you get to a running pyrpl application all the way from a conda installation, i.e. what commands do you use to generate the conda environment for pyrpl and how do you install pyrpl? 

Thanks!

ttuibk, 2018-07-27:
I'm having the same problem (win10).    I initially used the windows binary but decided to try the python version to see if I could get the most recent version.  I followed the instructions on the Github page for the master fork (including creating pyrpl-env).  After that, I did

conda update -n root conda
conda update --all

The suggested command to launch the gui did not seem to work.  I think maybe because my config file name has a space in it?
python -m pyrpl your_configuration_name

So instead I used the commands suggested in the executable cmd window.  These were entered in an IPython console (Spyder) or run as a script in Spyder:
from pyrpl import Pyrpl
p = Pyrpl('your_configuration_name')

EDIT:  I just did a test and now the scope issues disappear if I launch the gui without a configuration (and then click on cancel in the config file selection window) from the cmd window (pyrpl-env).  The scope issues reappear when I go back to launching from the IPython console.

closed 2026-09-23 by michaelcroquette