pyrpl-fpga/pyrpl #339: Trouble to make Pyrpl work with several redpitayas

HJacqmin, 2018-05-24:
Hello,

I am trying to open several Pyrpl GUI on the same computer (windows). Each GUI corresponds to one redpitaya. I use a local DHCP server assigning an IP address to each redpitaya. When starting a second GUI while the first one is already working, I have an error message :
[image]

For a better understanding of the bug, I used a Tera Term interface for each redpitaya. When starting the second one, I have this error : 
'skbuff: skb_over_panic'
'kernel BUG'
see : 
[image]

Thank you for helping !

HJ

lneuhaus, 2018-07-01:
Hi HJ, 

sorry for the delayed answer here. Is your problem still bothering you? Could you post how you start the two GUI's for the two redpitayas please? I.e.
- which commands do you use to get to that error message?
- are you using binaries or raw python code,
- which branch of pyrpl (master?)
- does each redpitaya work individually with pyrpl?

thanks!

HJacqmin, 2018-07-12:
Hi,

Thank you for your answer.

The 'kernel panic' error is solved. It was due to the duplication of the
file 'pyrpl-master'.

I still cannot start several GUI's using a local DHCP servor. I checked
that each redpitaya works individually with pyrpl in this configuration.
To start a GUI, I use the following command in a anaconda prompt :
> cd C:\Users\HJ\pyrpl-master
> activate pyrpl-env
> python -m pyrpl filename

I use raw python code. I get the error message when starting the second
GUI while a GUI is already runnning.

I plugged each RP on independant network cards to get around this issue
(to do so, I switched to static IP adressing). In this configuration, I
can start several GUI's. However, I really would like to make pyrpl work
with the DHCP servor to increase the number of RP...

Thanks !

HJ

lneuhaus, 2018-07-13:
So now, as you can make it work when different redpitayas are on different networks with different static IPs, we can narrow the error to one with IP adresses/networking issues. 

I propose you try first to ssh to all the redpitayas connected to your DHCP. If this already does not work, it is not a pyrpl issue but a problem with your DHCP settings.

We have operated 5-10 redpitayas on the same LAN with a DHCH server, but we used fixed IPs (based on a mapping between IPs and each Redpitaya's MAC address in the dhcp-config-file of the DHCP server). Also on a home network, you can enter a fixed IP for each MAC address in the Router settings. Let me know how this goes.

HJacqmin, 2018-07-19:
I switched to fixed IPs and plugged the redpitayas on an ethernet hub. I
performed the following tests :

1) windows, raw python code (via anaconda) :

When starting the second GUI, I get the same error. Note that an error
message appears on the first running GUI when starting the second one :

2) Linux (virtual box) :

I first tried to ssh the redpitayas, as suggested and it works :

Then, I first ran pyrpl from the binary files. I get the same error when
starting the second GUI.

I also installed the python source code and obtained the same error.

I performed serial monitoring of the second redpitaya while the first
one is running with pyrpl and obtained the following error message when
starting the second GUI :

lneuhaus, 2018-07-20:
Ok, thanks. Could you post again the error message you mention at the end of your post? It is not displayed so maybe it got lost on the way to github.

Also: how do you launch pyrpl for all these, i.e. how exactly do you pass them the ip address of the different redpitayas? 
If this turns out to be correct, I will need more information to debug your problem since in general, pyrpl has worked before with multiple instances running on different redpitayas. 
For that, you should edit the file global_config.yml in the directory pyrpl_user_dir under your home directory and set 'loglevel: debug'. Then, please copy-paste the entire text output from start (incl the commands) to where the error message appears. This is the only way ibsee to get closer tobunderstanding the problem here. Thx

HJacqmin, 2018-07-20:
Sorry for the missing error messages. Please find attached the required
data (procedure for IP adressing, error message, command lines...).

Redpitayas are connected to a ethernet hub.
Connection details of the network :
DHCP activated : no
Adress IPv4 192.168.1.7
Mask 255.255.0.0
NetBIOS on TCP/IP activated

1) Setting the IP adress of the redpitaya

	1a) Setting the proper IP adress

I plug the Micro-SD card from the redpitaya into my computer. I modify the 
files 'init.d' and 'interfaces' as follows : 

file 'rCs'in etc\init.d :
-------------------------------------------------------------
#!/bin/sh

# User startup script

# Uncomment the following and set a proper <DNS_IP> if your
# network configuration is static.
 echo "nameserver <192.168.1.102>" > /tmp/resolv.conf

echo "Feel free to add custom initialization commands here."
--------------------------------------------------------------

file 'interfaces' in etc\network :
--------------------------------------------------------------
# $Id: interfaces 955 2014-01-16 13:15:20Z ales.bardorfer $
#
# Red Pitaya network configuration
#

###########################
# lo: Loopback interface  #
###########################

auto lo
iface lo inet loopback

######################################
# eth0: Wired Ethernet - 1000Base-T  #
######################################
#
# Uncomment only one: dynamic or static configuration.
#

# Dynamic (DHCP) IP address
#iface eth0 inet dhcp
#    udhcpc_opts -t7 -T3

# Static IP address
# Don't forget to set the static DNS IP in sd:/etc/init.d/rcS too.
iface eth0 inet static
    address 192.168.1.102
    netmask 255.255.255.0
    gateway 192.168.1.1

################################
# wlan0: Wireless USB adapter  #
################################

auto wlan0
iface wlan0 inet dhcp
    pre-up wpa_supplicant -B -D wext -i wlan0 -c /opt/etc/network/wpa_supplicant.conf
    post-down killall -q wpa_supplicant
[... 353 lines cut ...]
  File "/usr/local/lib/python2.7/dist-packages/pyrpl/redpitaya.py", line 192, in __init__
    self.start_ssh()
  File "/usr/local/lib/python2.7/dist-packages/pyrpl/redpitaya.py", line 234, in start_ssh
    return self.start_ssh(attempt=attempt+1)
  File "/usr/local/lib/python2.7/dist-packages/pyrpl/redpitaya.py", line 234, in start_ssh
    return self.start_ssh(attempt=attempt+1)
  File "/usr/local/lib/python2.7/dist-packages/pyrpl/redpitaya.py", line 234, in start_ssh
    return self.start_ssh(attempt=attempt+1)
  File "/usr/local/lib/python2.7/dist-packages/pyrpl/redpitaya.py", line 249, in start_ssh
    e))
pyrpl.errors.ExpectedPyrplError: 
Could not connect to the Red Pitaya device with the following parameters: 

	hostname: 192.168.1.101
	ssh port: 22
	username: root
	password: ****

Please confirm that the device is reachable by typing its hostname/ip address into a web browser and checking that a page is displayed. 

Error message: timed out

        An expected error occured in PyRPL. Please follow the instructions
        in this error message and retry! 
----------------------------------------------------------------------------------------------------

lneuhaus, 2018-07-20:
Ok thanks. Just to be very clear: what IPs do you assign to your 2 Redpitaya's?

HJacqmin, 2018-07-20:
redpitaya1 : 192.168.1.100

redpitaya2 : 192.168.1.101

Thanks

HJacqmin, 2018-07-20:
Just a precision : I have a third redpitaya : 192.168.1.102 (this is why
this IP may appear in the files I send you).

lneuhaus, 2018-07-20:
And you confirm that if you do it the other way around (first connect to redpitaya with IP address 192.168.1.101, and second connect to 192.168.1.100, you get the same error on the redpitaya with IP 192.168.1.100? 

If that is so, I have no clear idea where the error comes from. But since you have some kernel error message, it seems like you have serious problems with the OS running on your redpitaya. My only suggestion in this case is to try a different version - unfortunately the setting up of static IPs may work differently on that version. This is why I prefer using DHCP with IPs assigned as a function of the MAC address - this works equally well with any Redpitaya OS. What version of the OS are you running btw?

HJacqmin, 2018-07-24:
I confirm that I get the same error on the redpitaya with IP
192.168.1.100 when I first connect the redpitaya with IP adress
192.168.1.101 and second connect the redpitaya with IP adress 192.168.1.100.

The version of the OS is 0.92-378.

I converted the redpitayas to dynamic IPs and used DHCP with IPs
assigned as a function of the MAC address. I got again the error (see
below).

New tests using DHCP with IPs assigned as a function of the MAC address :

1) Windows

1a) Prompt anaconda 1 - redpitaya 1
-------------------------------------------------------------------------------

(base) C:\Users\ONERA>cd C:\Users\ONERA\pyrpl-master

(base) C:\Users\ONERA\pyrpl-master>activate pyrpl-env

(pyrpl-env) C:\Users\ONERA\pyrpl-master>python -m pyrpl redpitaya1bis
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no
 options!
0 C:\Users\ONERA\pyrpl-master\pyrpl\__main__.py
1 redpitaya1bis
Calling Pyrpl(**{'config': 'redpitaya1bis'})
DEBUG:pyrpl.memory:Loading config file C:\Users\ONERA\pyrpl_user_dir\config\redp
itaya1bis.yml
INFO:pyrpl:All your PyRPL settings will be saved to the config file
    C:\Users\ONERA\pyrpl_user_dir\config\redpitaya1bis.yml
If you would like to restart PyRPL with these settings, type "pyrpl.exe redpitay
a1bis" in a windows terminal or
    from pyrpl import Pyrpl
    p = Pyrpl('redpitaya1bis')
in a python terminal.
INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname 192.168.1
.100.
QLayout::addChildLayout: layout "" already has a parent
ERROR:pyrpl.redpitaya_client:Error occured in reading attempt 0. Reconnecting at
 addr 0x40110000 to _reads value 16384 by client 1
---------------------------------------------------------------------------------

1b) Prompt anaconda 2 - redpitaya 2

---------------------------------------------------------------------------------

(base) C:\Users\ONERA>cd C:\Users\ONERA\pyrpl-master

(base) C:\Users\ONERA\pyrpl-master>activate pyrpl-env
[... 344 lines cut ...]
[TFTP-SETTINGS]
EnableTFTP=0
ROOT=C:\Users\LASBE\redpitaya\dhcpsrv\wwwroot
WritePermission=0

[HTTP-SETTINGS]
EnableHTTP=1
ROOT=C:\Users\LASBE\redpitaya\dhcpsrv\wwwroot

[00-26-32-F0-60-B6]
IPADDR=192.168.1.101
AutoConfig=07/23/2018 16:12:16
LeaseEnd=1563553502
[00-26-32-F0-60-22]
IPADDR=192.168.1.102
AutoConfig=07/23/2018 16:12:17
LeaseEnd=1563553505
[00-26-32-F0-59-90]
IPADDR=192.168.1.100
AutoConfig=07/23/2018 16:12:17
LeaseEnd=1563553500

4) Version OS

Red Pitaya GNU/Linux/Ecosystem version 0.92-378

michaelcroquette, 2026-10-01:
Closing as a duplicate of #362, which tracks running several PyRPL instances / Red Pitayas and the server kill-and-reconnect loop. Please continue there.

closed 2026-10-01 by michaelcroquette