pyrpl-fpga/pyrpl #341: pyrpl GUI long time running arise ConnectionAbortedError: [WinError 10053] ....OSError: Socket is closed

engeer, 2018-07-24:
Hi !
    When i running a pyrpl GUI in my computer with win10. To begin with everything work well, but running about two hours, the pyrpl arise the fellow problem:
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "pyrpl\redpitaya_client.py", line 154, in try_n_times
  File "pyrpl\redpitaya_client.py", line 113, in _reads
ConnectionAbortedError: [WinError 10053] 你的主机中的软件中止了一个已建立的连接。

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "pyrpl\widgets\module_widgets\lockbox_widget.py", line 970, in update_lockstatus
  File "pyrpl\widgets\module_widgets\lockbox_widget.py", line 987, in _is_locked_display_color
  File "pyrpl\software_modules\lockbox\lockbox.py", line 402, in is_locked
  File "pyrpl\software_modules\lockbox\output.py", line 129, in is_saturated
  File "pyrpl\attributes.py", line 148, in __get__
  File "pyrpl\hardware_modules\pid.py", line 154, in get_value
  File "pyrpl\modules.py", line 746, in _read
  File "pyrpl\modules.py", line 740, in _reads
  File "pyrpl\redpitaya_client.py", line 97, in reads
  File "pyrpl\redpitaya_client.py", line 165, in try_n_times
  File "pyrpl\redpitaya_client.py", line 172, in restart
  File "pyrpl\redpitaya.py", line 447, in restartserver
  File "pyrpl\redpitaya.py", line 385, in startserver
  File "pyrpl\redpitaya.py", line 406, in endserver
  File "pyrpl\sshshell.py", line 89, in ask
  File "pyrpl\sshshell.py", line 84, in askraw
  File "pyrpl\sshshell.py", line 63, in write
  File "site-packages\paramiko\channel.py", line 715, in send
  File "site-packages\paramiko\channel.py", line 1081, in _send
OSError: Socket is closed
ERROR:pyrpl.redpitaya_client:Error occured in reading attempt 0. Reconnecting at addr 0x40320100 to _reads value 1 by client 1
ERROR:pyrpl.redpitaya:Server not responding...
Traceback (most recent call last):
  File "pyrpl\software_modules\loop.py", line 74, in main_loop
TypeError: _lockstatus() missing 1 required positional argument: 'self'
  
What caused this problem? and How should I solve this problem?
Thanks for your reply!

 Jim
2018.7.24

lneuhaus, 2018-07-24:
Hi Jim, 

what seems to happen here is that after a while, the connection between your Pyrpl client and the redpitaya gets broken (this is what [WinError 10053]  means). Pyrpl uses an ssh (port 22) and a TCP data connection (default port 2222), and the traceback shows that both connections are broken. 
How do you recover from the error? Does a pyrpl restart suffice or do you have to reset your network router or redpitaya? 
If a simple restart is enough, one could think about fixing this with a more robust pyrpl version (that allows more disconnections before raising errors). However, I would prefer (and it would be much faster) if you would first attempt to solve this problem by modifying your LAN, because imho this error shows that your LAN connection is unstable. How well can you reproduce the two-hour interval for this error? If it is very precise, it gives us the hint that there is a timeout or firewall somewhere that closes connections after 2h or so. If it is rather random, it shows that the network is simply unstable. 

So after confirming the interval reproducibility, maybe you can check your networks firewall and timeout settings, confirm that all cables and power supplies are stable, and maybe even try with a different computer+LAN router or smaller LAN if you can get rid of that bug. 

The more information you give here, the more I can try to help you. If none of this works, and in the longer term anyways, we can take this issue as a motivation to make Pyrpl more robust w.r.t. network disturbances. 

Also, we should think of ways to include unstable network conditions in our unittests. 

PS: I have been running pyrpl instances for several weeks, with continuous logging transferring data forth and back between the Pyrpl client and the redpitaya. For that we had a separate LAN only for the devices in the lab.

engeer, 2018-07-25:
Hi Leo!
      Thanks for your patient and careful reply，i have solved this problem. The scpi manager need to be opended on the red pitaya and the method is use browser  log to the redpitaya，then open  scpi manager
as the picture show, the problem is solved, although i don't now why.
[image]

And i have another question about pyrpl's lockbox, what is the feedback bandwidth of lockbox？ can you give me a more details document of the lockbox? because when i use lockbox module, i don't know some of parameter meaning.

Thank you again for your reply！

Best regards

Jim
2018.7.25

lneuhaus, 2018-10-15:
feedback bandwith is 50 MHz, limited by the analog anti-aliasing filter at the inputs and outputs of the redpitaya. But you can place digital filters to lower that bandwidth. Latency from input to output of about 150 ns allows for a maximum locking bandwith of a few MHz.

closed 2018-10-15 by lneuhaus