pyrpl-fpga/pyrpl #343: ASG output not working

ecdlguy, 2018-09-18:
Hi, 
when switching on the ASG output, I get the following error:
```
INFO:pyrpl.modules:Calling setup() for asg0.trigger_source ...
ERROR:pyrpl.redpitaya_client:Error occured in reading attempt 0. Reconnecting at addr 0x40210000 to _writes value [    0     3     6 ... 16375 16378 16381] by client 1
```
This happens with the linux binary version (both "stable" and developement) as well as with the pip install.
The Red Pitaya runs on 
```
# Red Pitaya GNU/Linux Ecosystem
# Version: 0.98
# Build: 685
# Branch: 
# Commit: 0759d718c2f98bed50c2872a4656695710d6357a
# U-Boot: "redpitaya-v2016.4"
# Linux Kernel: "branch-redpitaya-v2017.2"
# Pro Applications:  b50f2409fddc9092c307e0fdde4ae2f833e0ab72 Applications (v0.97-RC6-7-gb50f240)
```

Using the windows binary it works. 

cheers,
Thorsten

asdfawer2, 2019-03-21:
Hi,

I do on ubuntu 18.04 get exactly the same error.

SilasRo, 2019-07-26:
Hello there, 
since this post a few month are past ... 

I had the same issue: #379 

In our case we solved the problem by installing a new network driver: 
In our PC is a RTL 8111/8168/8411 Ethernet controller from Realtek. But in linux automatically the driver r8169 is installed. 
You can see these infos by typing into the terminal: sudo lshw -class network 

Following the description of this website: 
[](https://unixblogger.com/how-to-get-your-realtek-rtl8111rtl8168-working-updated-guide/)
the Realtek r8168 driver can be installed. 

This solved at least this problem for me. 
regards, 
Silas

lneuhaus, 2019-07-26:
https://github.com/lneuhaus/pyrpl/commit/34a3c12284d64dc343d48f909110556391f312c3 fixes thus, will be merged to master soo n