pyrpl-fpga/pyrpl #348: Implement simple error signal free locking scheme

ecdlguy, 2018-11-19:
Hi,

I just stumbled across [this](https://aip.scitation.org/doi/full/10.1063/1.4903869) paper and I was wondering if it was possible to implement the algorithm directly in the FPGA code. The benefit is that it allows locking without the need for a modulation (PDH) or polarization spectroscopy of the cavity.

cheers,
Thorsten

lneuhaus, 2018-11-20:
Hi Thorsten, 

the algorithm in that paper is quite simple and should take little effort and few FPGA resources to implement. I think a good place to put it would be the PID module, as it provides the infrastructure (input + output) and is the place to look for locking functionality. Then, one could switch the output signal from an ordinary PID signal to the output of this minimization algorithm. 

I am not fully decided yet whether the algorithm is actually useful. The advantage is - as you say - less setup time as there are less tunable parameters such as modulation frequency and the like. The disadvantage is in my opinion that there will nevertheless be an oscillation (while continuously searching for the minimum error signal, the algorithm will "bounce" between the left and right slope of the minimum, with a frequency distribution and amplitude that should depend on signal amplitude vs output step size and update frequency of the algorithm, delay of the piezo response, and the level of noise of the input signal), and this oscillation will occur at a badly defined frequency, as opposed to the well-defined modulation peak of PDH. 

My conclusion:
1) If you simply want to avoid a phase modulator and dont like the Haensch-Couillaud-scheme, I would simply recomment a "synchronous detection", which is similar to a PDH but with much lower modulation frequency (one that can be generated with the cavity piezo). 
2) One should test above algorithm in more practical cases, and maybe even characterize the modulation peaks in more detail to know what to expect ahead of time. This will essentially result in further developing above algorithm, for example to automatically select the best step size and update rate, etc. I do not have time for this right now, but if you want to do the testing/finetuning, I can implement the algorithm for you if you want.
3) The algorithm might be very useful to approach a resonance without requiring additional parameters, before switching to a better lock quality with another error signal. But this requires minimum testing before.

Let me know if you want to pursue direction 2 in more detail.

ecdlguy, 2018-12-11:
Hi Leonhard,

thanks for digging into this! 

Of course one could implement a simple top-of-fringe lock with a small sine modulation of the laser frequency, but I think the algorithm is a bit different, since the amplitude "modulation" is changed dynamically. Also, there's no well defined frequency as you've mentioned.

It would be nice to try it and compare it's performance to PDH. So I'd be happy if you find the time to implement it, but since it's more or less a playground, please assign lowest priority on that one.

- Something totally different: Do you guys know why it's possible to use the PDH modulation at, say 20 MHz, to lock a second, subsequent cavity? In my picture, the sidebands cannot survive the first cavity (should be reflected, not transmitted) because its finesse is to high. 

cheers,
Thorsten