pyrpl-fpga/pyrpl #349: Async acquisition broken in python 3.7

SamuelDeleglise, 2018-11-19:
The function ` _wait_for_done(self, timeout)` in async_utils.py is no longer working properly with python 3.7. This is a quite critical problem since for instance this causes the scope to never refresh in normal mode.

The reason is not entirely clear to me, however, it s probably related to some changes in asyncio in python 3.7 that makes the Future objects bounded to a particular event loop, such that the callbak that was registered with `add_done_callback(self._exit_loop)` never gets called when executing `self.loop.exec_()`

I am working on a quick fix that should be available soon in develop 0.9.3, however, in my opinion, all the support for asynchronous acquisition could become 40 dB more readable by using the framework introduced in python 3 with async/await ... The main reason why we still have these low-level Future and callback mess is for the code to stay compatible with python 2. For this reason, I would strongly recommend giving up support for python 2...

lneuhaus, 2018-11-20:
Hi Samuel,
I finally agree with you that we can sacrifice python 2 support in exchange for more readable code. I propose the following procedure:
1) Keep your quick fix in develop-0.9.3 for now. 
2) Let me fix the other issues with the develop branch (which are the reason why I have not pulled it into master yet. Once the branch is ready, we will release a new version, which will be tagged as the final version with python 2.7 support.
3) In the meantime or after, we can drop the python 2 support in whatever comes after develop-0.9.3. If you want to start working on that, I suggest using a new branch "python3-only" derived from develop-0.9.3.

Do you agree with this?

SamuelDeleglise, 2018-11-21:
OK sounds great,
I will start the pyton3-only as soon as I have some spare time... lots of garbage code should go away then...

lazyoracle, 2018-11-29:
I get the following error when using `scope.curve_async( ).result( )` - 
` async_utils.py in result(self)
    132         try: #  concurrent.futures.Future (python 2)
--> 133             return super(PyrplFuture, self).result(timeout=0)
    134         except TypeError: #  asyncio.Future (python 3)
TypeError: result() takes no keyword arguments

During handling of the above exception, another exception occurred:

InvalidStateError                         Traceback (most recent call last)
<ipython-input-13-90941532cfd6> in <module>
     55 # The function curve_async returns a *future* (or promise) of the curve. To
     56 # access the actual curve, use result()
---> 57 ch1, ch2 = res.result()
     58 
     59 # plot the data

C:\Users\Anurag\AppData\Local\conda\conda\envs\pyrpl36-env\lib\site-packages\pyrpl\async_utils.py in result(self)
    133             return super(PyrplFuture, self).result(timeout=0)
    134         except TypeError: #  asyncio.Future (python 3)
--> 135             return super(PyrplFuture, self).result()
    136 
    137     def _exit_loop(self, x=None):

InvalidStateError: Result is not set. `

Is this part of the same problem?

SamuelDeleglise, 2018-11-29:
I don't think that is a bug...
If you are trying to retrieve the result of a curve that is not yet acquired, you expect this error to happen.
On the other hand, if you want the function call to block until the result is ready, you should use `scope.curve_async().await_result( )`
or equivalently, use the synchronous function provided for that purpose:
`scope.curve()`

lazyoracle, 2018-11-29:
> I don't think that is a bug...
> If you are trying to retrieve the result of a curve that is not yet acquired, you expect this error to happen.
> On the other hand, if you want the function call to [...]

I am trying to follow the exact steps mentioned [here](https://pyrpl.readthedocs.io/en/latest/api.html#module-pyrpl.hardware_modules.scope) in the doc to reproduce it. Am I missing something?

lazyoracle, 2018-11-29:
> OK, it looks like there is indeed a bug: the curve never gets ready until the jupyter cell returns. This is definitely related to the issue we are discussing here. Just to make sure, are you using the branch master or develop-0.9.3 ?

I'm on the master branch. Should I try with dev-093? Have you already committed some fixes to this there?

lneuhaus, 2018-11-29:
Yes, Samuels fix is on the develop-0.9.3 branch, commit 9024954. Either you switch to develop-0.9.3, or you just cherry-pick the one commit into your local master branch. There is a lot of different stuff in develop-0.9.3, which is why I have not yet merged it into master..

SamuelDeleglise, 2018-11-29:
Actually, the bug you have spotted is in both branches... 
This section of the notebook was apparently not tested in a long time, because it cannot work this way anymore:
the function `time.sleep()` doesn't allow the event loop to retrieve the curve from the redpitaya when it is ready because no other event can be treated while "sleeping".
The simplest fix I can think of is to replace 

`from time import sleep` 

by 

`from pyrpl.async_utils import sleep` 
 
in the notebook...

lazyoracle, 2018-11-30:
> 
> 

> The simplest fix I can think of is to replace
> 
> `from time import sleep`
> 
> by
> 
> `from pyrpl.async_utils import sleep`
> 
> in the notebook...

Thanks, this worked. Should I edit the docs and make a PR?

lneuhaus, 2018-12-01:
That would be cool, yes! Thanks in advance

SamuelDeleglise, 2018-12-01:
OK, the final fix is here:
https://github.com/lneuhaus/pyrpl/commit/a37d468c1976a7727bcdb9fc9a80485aaffe9f48
That was not many lines of code but testing on 2.7, 3.5, 3.6 and 3.7 on notebooks + unittests was quite lengthy...

SamuelDeleglise, 2018-12-02:
It passes unittests on all these versions and a quick test of the scope gui is also fine in the notebooks...

SamuelDeleglise, 2018-12-04:
OK, I have already advanced quite a bit on the python3-only branch. The code will be compatible with python 3.5, 3.6 and 3.7. Support for python <=3.4 has to be dropped since the syntax `async def some_coroutine()` was only introduced in python 3.5.

The main difficulty is still to find a way to deal with asynchronous programming that integrates nicely in Ipython kernel (with a qt event loop) in all 3 python versions.  The code to perform that is located in a new version of the file async_utils.py: This module provides a function `wait(future_or_coroutine)` that will stall untill the coroutine is finished (while not blocking the event loop execution). In order to work, the futures need to be sent to the function execution via a custom version of `ensure_future` and use a custom version of `sleep` that are provided in the same file. In practice, they simply execute in a dedicated `QEventLoop` named `LOOP` that can be waited for by running `LOOP.run_until_complete(some_coroutine)` (python 3.7), or use the old trick of creating a new `QEventLoop` on the fly and executing it via `loop.exec_()` (python 3.5 and 3.6).

With that, all the mess in acquisition_modules.py, scope.py, spectrum_analyzer.py, network_analyzer.py can be replaced by some highly readable coroutines, but very little change will occur in the interface:

```
single_async() # launches the acquisition of trace_average traces and returns immediately
single()      # idem but blocks until the curve is ready
continuous()  # launches an infinite execution and returns immediately.
```

For instance, the code for single_async is as clear as:
```
    async def _single_async(self):
        self.running_state = 'running_single'
        self._prepare_averaging() # initializes the table self.data_avg and self.current_avg
        for self.current_avg in range(1, self.trace_average + 1):
            self.data_avg = (self.data_avg * (self.current_avg-1) + \
                             await self._trace_async(0)) / self.current_avg
            self._emit_signal_by_name('display_curve', [self.data_x,
                                                        self.data_avg])
        self.running_state = 'stopped'
        return self.data_avg

```
Note that all the beauty is in the `await self._trace_async(0)`  that gives back the control to the eventloop untill the result of the other coroutine `self._trace_async(0)` becomes ready. The big advantage over the Timer/callback approach used previously is that all the code for the successive trace acquisitions is now written sequentially in one single place.

The main change that I plan to introduce in the interface to make it more consistent is the following:
I have made `curve_async` a private function (and renamed it `_trace_async`) and removed `curve` since the behavior was essentially the same as `single` with `trace_average=1` (except for the fact that gui was not updated, but that just makes things confusing I believe: if one doesn't want to update gui, he should either turn it off or ask for it explicitly somewhere)

SamuelDeleglise, 2018-12-04:
I also suggests that `running_state` should be a "read-only property" since it is misleading to load the state of an instrument that is running for instance... 
To launch an instrument, one would have to explicitly call the functions `single`, `single_async `or `continuous`
This caused a lot of trouble to debug in the unittests for instance when we found out some instruments where acquiring continuously in the background...

lneuhaus, 2018-12-05:
I think there is no problem in running multiple lockbox modules in a single pyrpl instance (analogous to multiple NAs etc), given there are enough resources. We would just to make a button for that i think. But it is probably a good idea to verify that. 

I also think to remember running multiple pyrpl instances in a single kernel without fundamental problems, e.g. one as a property of another one's lockbox. 

Is there a problem to have two GUIs in the same kernel? The only problem i see here that e.g. the scope framerate will be halved. To fix this, we would need to make the read/write functions of the client asynchronous, which is a change with the risk to add a lot of problems. I will add a new issue for that discusion, ok?

lneuhaus, 2018-12-05:
I just did a few tests:
1) Having several lockboxes in one pyrpl instance is possible. But we should have a defaualt lockbox with only 1 output, as otherwise there are not enough PIDs for two lockboxes in the default case. 

2) Running several scopes in parallel on a single kernel is possible, but on my computer it reduces the framerate rougly in proportion to the number of instances. I did the test with hostname='_FAKE_' which needs about 140 ms on my computer to read both scope channels. So it remains true that we might gain in framerate by not wasting so much time on waiting for the scope data to arrive, but there are clearly limits, as a quarter of the time is simply spent on signal processing with numpy. Here is my test code
```
import pyrpl
import time

# make multiple instances of pyrpl, start the guis
ps = [pyrpl.Pyrpl(config='my_test_%d'%i, hostname='_FAKE_') for i in range(4)]
for p in ps:
    s = p.rp.scope
    s.continuous()

# compute the frame rate for the last pyrpl instance
def display_frame_rate(*args, **kwargs):
    if not hasattr(s, "_count"):
        s._starttime=time.time()
        s._count = 0
    else:
        s._count += 1
        fps = s._count / (time.time() - s._starttime)
        if s._count == 20:
            s._logger.info('FPS: %.2f', fps)
s._signal_launcher.display_curve.connect(display_frame_rate)

# 1 instances * 5.13 fps = 5.13 fps
# 2 instances * 2.85 fps = 5.70 fps
# 3 instances * 2.09 fps = 6.27 fps
# 4 instances * 1.50 fps = 6.00 fps
```

lneuhaus, 2018-12-05:
By the way, thanks for the changes in the new python-3 branch, I agree with most of the API changes. 

The only weird thing is to have that private setup_attribute. But for the moment I have no better idea to solve the problem.

SamuelDeleglise, 2018-12-06:
Yes, I went through various possibilities until I went back to this one... Maybe a slightly cleaner option would be to take the private attribute `_running_state` out of the setup_attributes and only keep a boolean flag `run_continuous` in the setup attributes. The `running_state` (without underscore) could then be exposed as a read-only property.

The idea being that only the 2 working states `running_continuous` and `stopped` can be saved and restored latter. All the rest should be implementation details...

SamuelDeleglise, 2018-12-08:
- OK I implemented the `run_continuous` boolean attribute as part of the _setup_attributes... I believe it makes more sense. 

- By the way, I have measured something like 20 fps for 1 scope and 10 fps for 2 scopes running in parallel in rolling mode on real hardware (keep in mind that they are limited in practice to 50 fps in rolling mode (I think it is even 25 in continuous normal mode) to avoid making the event_loop unresponsive).
I think the ratio benefit/effort for the asynchronous read/write is way  smaller than what I had in mind when starting looking at it (to be honest, I don't even have a clue how to attack the problem), so I put that on hold unless a new idea pops up.

- On the other hand, I now have all unittests OK on the branch python3-only (except for one with the IIR that seems to work only in python 3.5, the problem is also there in the other branches I think).  42a704d97eccf6baf5a02c343878e51ca2d2223d

- I basically replaced the LoopModule that is used in the lockbox by loops in coroutines. The only minor change to the user interface is `lock()` --> `lock_async()` for consistency since the function returns immediately.

lneuhaus, 2018-12-08:
1) Cool.
2) I agree. Lets take this up once there is a real need for it. It seems much easier to me to have two kernels running should this ever be a real problem. 
3) Does this mean the code is ready from your point of view? Once it is, don't hesitate to make a PR, and I can do a review on it.
4) Ill look into this when 3) is ready. Just for understanding: lock() still exists and is the sync-version of lock_async?

SamuelDeleglise, 2018-12-08:
I basically need to add some comments and help-strings. I will try to do that tonight and issue a pull request. But also, I am proposing below a few modification to the Lockbox API for consistency. Moreover, I realize that the unittetsts for these functionalities are quite limited, I will try to add some more unittests as well. 

lock_async(): # I propose to add the _async for consistency
      """
      Launches the full lock sequence and returns immediately.
      **Proposed return value**
      Returns a Future that evaluates to True if lock was successful, False otherwise
      """

lock(): # It looks like this function doesn't exist at the moment
        """
        Same as lock_async except it only returns a boolean at the end of the lock sequence
        """

sleep_while_locked(time_to_sleep): # keep it (should be programmed with an Event())
        """
        wait for time_to_sleep and returns True if no unlock occured.
        Otherwise, returns False as soon as unlock occurs.
        """
lock_until_locked_async(**kwds): # _async is there for consistency.
        """
         runs the full lock sequence as many times as necessary.
        **Proposed return value**
        Returns a Future that evaluates to True when the lock is acquired.

lock_until_locked(**kwargs): # should add a timeout
        """
        runs the full lock sequence as many times as necessary and 
        only returns once locked.
        **Proposed return value**
        True if lock was succesful before timeout
        False otherwise.
        """

#==================================================#
#     If I understand correctly, all the following RElock_blabla are equivalent to
#     if not self.is_locked_and_final():
#            self.lock_blabla
#    I would vote to remove these functions, since the two lines above are more explicit...
#==================================================# 
relock(**kwds): # If we keep it, should be renamed relock_async()
        """
        If not locked, runs the full lock sequence once. 
        **current behavior**
        Returns True if locked at the beginning, False otherwise
        **consistent behavior would be**
        Returns a future object that evaluates to True if locked at the end, False otherwise
        """
relock_until_locked_async(**kwds):
        """
        Same as lock_until_locked_async, but does nothing if already locked
        """

relock_until_locked(**kwds):
        """
        Same as lock_until_locked, but returns immediately if already locked ???
        """

SamuelDeleglise, 2018-12-10:
Two more things: 
 - I realized while debugging the python-3-only version that there was actually still a flaw in my bugfix for develop0.9.3. The function `_wait_for_done()` for python 3.7 was working fine unless called by a qt slot (for instance when you press calibrate() in the lockbox, there is a call to `scope.single()` that needs to wait for the curve to be ready). The work around is simply to use the same code than for python 3.6 and 3.5. I am not sure why I used different implementations in the first place. The only place where you need to treat python 3.7 separately is when you create the future, (async_utils.py line 123), where a separate loop needs to be specified. This is a bit black magic to be honest, but at least, the black magic is well confined within async_utils.py. Please, include the patch before making the final release of 0.9.3 (unittests ongoing...).
- I have added in the branch python-3-only a unittest for notebooks (to avoid this kinds of undetected problems). It basically fails if any of the notebook cell returns an exception. Also, this is checking that the output of the first cell is the same version number as the current python. This is to avoid testing an IPython kernel with the wrong version number. It would be nice to add the tutorial notebook in this unittest (at the moment it is testing all .ipynb located in the directory, but we could add a hardcoded linked to the tutorial).

lneuhaus, 2018-12-12:
The develop-0.9.3 release will have to wait for me having re-setup a working CI system imho. I am playing with running jenkins on my personal computer at the moment. 

How do you run the unittests at the moment? You have any kind of automatization for testing in different python versions? If not, I recomment to translate the travis-script into a command-line script. If you are using linux, I can send you mine.

SamuelDeleglise, 2018-12-12:
No, I have 3 different conda environments and I launch manually the unittests on one after the other. Also, on branch python-3-only, I have added a notebook test that I didn't put in develop-0.9.3. If I understand the philosophy, for anything that's not specific to python3-only development, you would like me to work on develop0.9.3 and only pull them on the branch python3-only afterwards instead of directly working on python3-only.

I am using anaconda under windows. I think that's also fine to just wait for your continuous integration system if you are confident that this will eventually converge.

closed 2026-09-23 by michaelcroquette