pyrpl-fpga/pyrpl #350: Sinchronyzing the phases of two iqs

michaelcroquette, 2018-11-30:
Hi,

We would like to lock two cavities with one red pitaya. We use one output of the redpitaya to send a 20 MHz modulation to an EOM that phase modulates the laser beam that goes through the two cavities and then lock the cavities with this modulation. The issue is that when we demodulate the two iqs have a different phase between them that changes every time we load pyrpl. Is their a way to synchronize the two phases or do we have to change the verilock code ?

lneuhaus, 2018-11-30:
Hi, 

there is a new branch https://github.com/lneuhaus/pyrpl/tree/syncronize-iqs that fixes this feature-request. After setting the frequecies to 20 MHz, call from any iq-module iq
`iq.synchronize_iqs()` and the relative phase will be zero.

SamuelDeleglise, 2018-12-01:
Cool! Did you do that overnight?

lneuhaus, 2018-12-01:
Yes :smile: but part of it was already there. Can you think of other modules that require synchronization?

Still to do: 
-  add synchronization call at iq module setup
- add synchronization button to gui
- fix unit tests (locally there is no problem, but travis keeps on failing due to connection issues)
- modify docs

Ill try to fix develop-093 asap so we can merge all these individual fixes together.

michaelcroquette, 2018-12-03:
Ok it's working great. Now we're wondering if we should pull it to develop 0.9.3 or make our own branch ?

Regarding other possible improvements we're already thinking about the next step which would be synchronizing several iqs accross physically separated redpitayas (To lock 3 cavities using only one coherent excitation). For this to work, we will need:

1/ to synchronize the clock signals of the 2 boards (I heard that it is possible with some physical connections)

2/ The easy way to implement the synchronisation would be to transmit a trigger signal accross the boards (for instance via one of the digital IO of the boards). Hence, it would be nice if the iqs would have a "trigger input" that can be configured. 

That's not ultra critical at the moment (we can use a tilt locking technique instead), but it's probably a quite general improvement that many people could benefit from).

lneuhaus, 2018-12-03:
I have simply pulled develop-0.9.3 into the synchronize-iqs branch in oder to keep debugging develop-0.9.3 as simple as possible. There was only a trivial conflict in modules.py. So I suggest you work with the synchronize-iqs branch for now and pull future changes from develop-0.9.3 into that branch.  

There should be a synchronize-iqs-call at startup and a gui button as well now. My latest commit to synchronize-iqs contains a recompiled fpga bitfile with develop-0.9.3 and synchronize-iqs changes together, but it seems there was no bitfile difference. So I am rather confident that your local fix is identical with mine. 
Once I will have had some time for fixing the develop-0.9.3 branch, I will merge everything together into the develop-0.9.3 and then the master branch. 

To your new feature request: This is indeed an upgrade that many people want, me included. It requires a bit of time, which I do not have right now but hopefully soon. If you want to start working on this, you are most welcome. 

For implementation, there are a number of possibilities:
1) https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extADC.html explains how you can in principle replace the crystal oscillator by an external clock. To be supplemented by a trigger signal from the external DIO's. 
2) https://www.koheron.com/blog/2016/11/29/red-pitaya-cluster.html which is simpler to implement, but might come with higher clock jitter on the ADC. Can be supplemented by a trigger channel on the STAT connectors. 
3) Some custom bus implemented on the extension connectors DIO pins, which is similar to 2, but has more wires at the cost of more work to get it running. My favourite, maybe even together with 2). 

The best would be to directly implement a maximum number of wires for Redpitaya interconnection, since one will sooner or later want to share digitized analog signals among different redpitayas at the highest possible rate. 

Sooner or later, this will be implemented!

lneuhaus, 2018-12-09:
I think it is a good idea to use the synch register to also enable freezing the integrators (or even all gains) of the pid modules. That is, as long as the sync register bit for a PID module is set, either the I-, the P- or both gains are zero. This would allow to change the gains of multiple modules quasi-simultaneously. A longer-term upgrade would enable to set any set of registers in the same clock cycle, but that could be overkill w.r.t. the resourcecost-functionality-tradeoff.

lneuhaus, 2018-12-10:
all features discussed here are implemented in develop-0.9.3 now, with the exception of synchronizing iqs over different boards (see #357).

closed 2018-12-10 by lneuhaus

michaelcroquette, 2018-12-10:
We don't see the synchronize_iq() function in develop0.9.3. Did you push your changes to origin ?

lneuhaus, 2018-12-11:
sorry, forgot to push. it should be online by now, along with a few other bugfixes/features.

lneuhaus, 2018-12-11:
Any objections if I delete the syncronize_iqs branch now?

SamuelDeleglise, 2018-12-11:
No I would say it's fine

ram-bhaskara, 2022-10-20:
@lneuhaus @michaelcroquette @SamuelDeleglise this issue is particularly relevant to what I want to do as well. In short, I want to use iq demodulation for phase measurement with my RP board synchronized with 10 MHz reference clock from, say, a function generator. Could you make your exercise a little more instructional for me to follow how to do this? 
I find https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extADC.html not operational and so are many RP docs. It is difficult for me to decode how to use the iqs module as well. Please help.

SamuelDeleglise, 2022-10-20:
What we have already done so far (by ascending order of difficulty): 
- synchronize several IQs on one redpitaya (there's even a function and button in the gui now)
- replacing the 125 MHz clock of the board by an externally generated 125 MHz clock. This is used to synchronize several redpitayas together
- In addition to the previous trick, which is used to sync the frequencies, use a digital trigger to also synchronize the phase of IQs on different redpitaya board

We have not yet tried to synchronize the boards with a 10 MHz reference (in fact our 125 MHz is generated thanks to some cheap evaluation board, such that it is not to constraining), all this will probably be documented in @michaelcroquette's thesis, stay tuned.

ram-bhaskara, 2022-10-21:
> What we have already done so far (by ascending order of difficulty):
>
> * synchronize several IQs on one redpitaya (there's even a function and button in the gui now)
> * replacing the 125 MHz clock of [...]

@SamuelDeleglise thank you! 

- What is the GUI you're talking about? I'm using the one launched via pyrpl-windows.exe and I fail to see a function button. I see iq0, iq1, iq2, iq2_2 as the four IQs
- Now, I'm trying to generate out1 (trigger immediately) via asg0 module. A sine wave 1kHz with amplitude 1, for example. I could see this output signal on the scopes module. I connected out1 to in1 on the RP using a bnc cable.
- Observation 1: I notice the sine wave at the in1 from about -0.9 to +0.9 (not same as I generated from out1). Frequency matches. 
-  Using iq0: with in1 as input, at frequency 1 kHz, bandwidth -18.97 to + 18.97, gain 10. I see a sine wave of frequency on an oscilloscope when I extract this signal to output_direct _out2_. The same output is delivered for _output_signal_ of types 'quadrature', 'pfd', 'output_direct','quadrature_hf'. What really is happening here? I expect the iq module output to deliver a mixed double frequency. And how can I extract phase information from iq demodulation from here? 

 Thanks in advance. Please help!

SamuelDeleglise, 2022-10-21:
I am talking about this one... I think the exe file is lagging far behind the current development.... Use the branch python3_only instead...

[image]

SamuelDeleglise, 2022-10-21:
> @SamuelDeleglise thank you!
>
> * What is the GUI you're talking about? I'm using the one launched via pyrpl-windows.exe and I fail to see a function button. I see iq0, iq1, iq2, iq2_2 as the four IQs [...]
OK, that might be due to imprecision in the analog electronics or attenuation in the cable, I am not too worried about that.

> * Using iq0: with in1 as input, at frequency 1 kHz, bandwidth -18.97 to + 18.97, gain 10. I see a sine wave of frequency on an oscilloscope when I extract this signal to output_direct _out2_. The [...]

There's a misunderstanding about how the IQ module works (and more generally all hardware modules featuring both "outpu_direct" and "output").... Here is the block diagram for the IQ module... "output_direct" is one output that can only be routed to the sumators of out1 and out2 (analog outputs of the redpitaya). "output" is the DSP output of the module, that can be sent to another module such as the scope or pid module for instance. The multiplexer where you get to choose what is routed to the output signal is only behind the "output" signal, not the "output_direct". Also, there's probably another misunderstanding when you are writing" bandwidth -18.97 to + 18.97, gain 10" a positive number in the bandwidth means a low pass filter, a negative number, a high pass filter. Two numbers!=0 mean that you are using a second order filter... In this case you are using a high pass and a low pass in series.... This is very weird.

[image]

> 
> Thanks in advance. Please help!

ram-bhaskara, 2022-10-22:
@SamuelDeleglise really appreciate your help here. I surely got some important info to begin with. 

> Finally, if you look closely the block diagram below, what you would like to set to non-zero is probably not the "gain" register but rather the "quadrature_factor", with the multiplexer set to "quadrature" and low pass filters set to 18 Hz (or [18 HZ, 18 Hz] if you prefer second order low pass).... 
> 

Here is what I have tried:
- Generated sine (1 kHz) using asg0, with output_direct = out1, I'd connect out1 to in1 via a BNC cable
- iq0 module with input as in1.
  Frequency of iq set at 1 kHz, LP filter bandwidth ([2429, 2429] Hz) to monitor the mixed frequencies (2 kHz) output. I'm not sure how to set the _quadrature_factor_. I'm testing around with 1 to 10.
- setting output_signal = quadrature and output_direct = out2, I'm monitoring the out2 on a digital oscilloscope. Is this a right thing to do? 
- Observation: I do not see a 2 kHz response on the DSO even if I increase the bandwidth above 2 kHz. I only see a 1 kHz sine wave (scope snippet attached). Bandwidth below 100 Hz would still have the 1 kHz. Why is that? 
- I ultimately would like to compare the phase between my signals. How do I interpret output='pfd' response? 

> Then you should inspect the signal with the internal scope of pyrpl (set input1 to iq0 for instance)... If you really want to have the signal outputed in the analog domain, use a pid module, where you set input to iq0, p gain to 1, i=0, setpoint=0 and output_direct to out1.
>
 What does input=iq0 mean here? In future, I would like to perform iq demodulation on a signal that is input through in1 (ADC1). Doesn't output_direct = out2 in the iq0 module output the signal to out2 in analog? How is pid helping here?

> This was a short debugging to help you get started, but at some point you will have to read the doc I am afraid ;-)

Could you please point me to the docs you're referring to. I'm following https://pyrpl.readthedocs.io/en/latest/gui.html but I couldn't find the necessary implementation details. I'm sorry for elaborate questionnaire but this would set me up with using the RP properly. Thank you so much!

[image]

SamuelDeleglise, 2022-10-22:
I think the main thing you are missing is the basic architecture of DSP modules and the distinction between their "output" and their "output_direct". The relevant section of the doc is 
[Here](https://pyrpl.readthedocs.io/en/latest/basics.html) ...

Please, look carefully at the IQ module block diagram that I posted in this thread: there is no way to send directly the demodulated quadrature directly to an analog output because the module is not wired this way... On the other hand you can sent the demodulated quadrature to "output" THIS IS A DIFFERENT THING !
The output of a module is a digital signal living inside the FPGA that you can monitor via the internal scope or connect to another module. Don't you have the possibility to set p.rp.scope.input1 = iq0 ? This would connect the "output" (not output_direct beware) of iq0 to the input1 of the scope

ram-bhaskara, 2022-10-26:
> I think the main thing you are missing is the basic architecture of DSP modules and the distinction between their "output" and their "output_direct". The relevant section of the doc is [...]

Thank you very much @SamuelDeleglise! I believe I'm now a bit clear on how to use Pyrprl iq demodulation. Ultimately, I aim to measure phase difference between RP reference and in1 signals. Using appropriate filter bandwidth, I'm able to eliminate high-frequency component to see a LP-filtered quadrature signal at _output_ that is sensitive to the in1 phase variation. 

How do I interpret the phase information from iq0 output data? How do I use this iq demodulation module and build upon to compute atan(Q/I)? 
If this is not a relevant query here, I would be happy to seek your suggestions in a different issue.

michaelcroquette, 2024-07-15:
I made a discussion to conclude on our work on the subject, see #519.