pyrpl-fpga/pyrpl #353: Travis unit tests are frequent source of problems

lneuhaus, 2018-12-02:
We can discuss here how to solve the problem of providing unit test results for all commits that are creazed.

Current problems are mostly related to
- communication delay between us-based travis server and europe-based Redpitaya of the order of 100ms
- issues due to broken connection during testing, related to the above. 

The best solution would be to run the unit tests on a europe-based machine with direct access to a Redpitaya. 

Another option would be to locally run the tests before committing code changes and prove that they were successful, e.g. with a hash of the commit or sth alike. This would require a proper way to locally create an environment similar to the one of travis.

Another good option would be to find a europe-based ci Service Provider, or use our own AWS server. 

Ano

SamuelDeleglise, 2018-12-03:
But Leo, at the moment is there a redpitaya physically accessible for Travis? Some time ago it was at my place, but since I moved, I didn't reinstall the necessary hardware... Moreover, this was rather cumbersome since my internet provider doesn't provide a fixed IP address.

michaelcroquette, 2026-10-01:
Closing as part of a triage of long-inactive issues: Travis CI is no longer used. Tests now run in GitHub Actions. Hardware-in-the-loop CI can be discussed in a new issue.

closed 2026-10-01 by michaelcroquette