pyrpl-fpga/pyrpl #354: Make Module.reads-function asynchronous

lneuhaus, 2018-12-05:
Or find another solution to not waste so much kernel time waiting for the data to arrive from the red pitaya.

lneuhaus, 2018-12-05:
Issue #241 is the same as this one

SamuelDeleglise, 2018-12-06:
I have been looking (drowning) into that today. And it looks like it is going to be very tough to do, at least with an eventloop approach. The reason is that all the communication to the redpitaya is hidden behind attributes such that the asynchronous code should be well isolated in the getter of the attributes (There is no way to have the coroutine architecture propagate beyond that level). And hence, we should have a way to wait for the setting of the attribute to be done (at least on one given redpitaya) until returning from the attribute getter. This is typically done with `LOOP.run_until_complete(future)`. This is the way I do it in `async_utils.wait()`.

Hence, the natural solution would be  to call my function `async_utils.wait()` in the getter. I have been testing it and it works until you call the getter from within a coroutine (for instance the one that is used in `scope._trace_async()`). It looks like python is allergic to the nested event loop structure (https://bugs.python.org/issue22239).

This basically forces us to have only one big island of coroutines stacked with `await` calls all the way from the bottom to the top of the stack--> No async redpitaya communication hidden in attributes

Maybe another option would be to do the communication in separated threads (one per redpitaya) that would be waited by the ...

lneuhaus, 2018-12-08:
I agree. However, what we could think of is a way to group communication with the redpitaya into "atomic" units to save communication time in certain cases. So something like this
```
with iq.atomic() as iq:
    iq.amplitude=0.4
    iq.frequency=30
```
would be sent as a single TCP packet. Do you think there is a need for this?

SamuelDeleglise, 2018-12-09:
I see, this could probably improve the performance. For instance the scope has an update period of 40 ms, eventhough the read time for a curve is 9 ms. I would not do it before having a clear breakup of the time spent in every method though... have you already used profiling libraries?

lneuhaus, 2018-12-09:
Nope not yet