pyrpl-fpga/pyrpl #357: synchronizing iq modules of different redpitayas

lneuhaus, 2018-12-10:
Discussion started here: #350

alexandrejourneaux, 2020-12-03:
Hi,

We are working with Michael and Samuel on the synchronization of the IQs of two different RedPitayas. We opted for "option 1" amongst the different solutions you mentionned in issue #350, as it seemed to be the quickest one: we managed to move the resistors R25 and R26 to location R23, R24 as explained here in the doc: https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extADC.html . We are the using the same external ASG to feed the new external clock input of both RedPitayas.

This solution works fine and seems very stable, however we are dealing with an obvious arbitrary phase difference between the IQs that are on two different RPs. We are wondering what is the best solution to synchronize them.

We thought of adding a trigger source to the IQs (in the spirit of what already exists in the ASG modules), in order to trig them simultaneously _via_ the DIO_0 pins (wired together). We would like your opinion on the best way to do this. Shall we use the sync_i input of the iq_block module? I guess it would not be too hard to wire the sync_i input of all IQ modules to DIO_0 in the FPGA, and change the synchronize_iqs() method to a simple switch from 0 to 1 (or 1 to 0, depending on how sync_i is dealt with by the IQ module) of DIO_0. Therefore, for two RPs with wired DIO_0 pins, pressing the "Synchronize IQ" button on the first would synchronize all the IQs on both RPs without any arbitrary phase shift.

Thanks for your help,
Alexandre

mgrossmann-sketch, 2024-07-01:
Dear Alexandre,
I assume this issue is somehow closed, as stated in (https://doi.org/10.1063/5.0178481):
"We have already used this functionality to synchronize several Red Pitaya boards and maintain a stable phase relation across multiple instances of PyRPL. Detailed guidelines are provided elsewhere. [Thesis Michaël Croquette]"
However, the detailed guidelines are quite short in my understanding. The hardware layout given, but software wise the only tip is:
"We add a connector between two digital Input/Output of the RedPitayas, and when one of the RedPitays starts its IQ
module it will send a signal to the other so that their internal reference at Ω always have the same phase relationship."

This seems to be the solution Alexandre talked about. 
Is this software already implemented in the official Pyrpl version?

Any hints are appreciated.

Best regards,
Martin