pyrpl-fpga/pyrpl #358: Lockbox UX improvement

SamuelDeleglise, 2018-12-11:
The LockBox GUI would benefit from some improvements.

A the very least, the buttons should be brought to the left, such that you don't need to scroll all the time to the right to press sweep or lock. Better would be to hide lots of stuffs in submenus or dialogs such that the widget size doesn't require sliders everywhere to fit in the screen...

lneuhaus, 2018-12-11:
another suggestion is to add the following setpoint-units:
- relative_relfection
- relative_transmission
- absolute_transmission
- absolute_reflection

This requires a nonlinear unit conversion (currently impossible because unit conversion factors are properties),  but nothing infeasible.

lneuhaus, 2018-12-14:
I agree on this one, go ahead if you have time. Thanks!