pyrpl-fpga/pyrpl #373: Synchronize asgs

SamuelDeleglise, 2019-03-06:
I am not sure how this could be done: what I want is to output 2 sines at the same frequency on out1 and out2 and to use the start_phase register of one of the asgs to change the relative phase between out1 and out2. 

Maybe the trigger options of the asgs could allow to do that but I didn't understand what the various options mean (ext_positive_edge ? ext_negative_edge ? raw ? high ?).

Unfortunately, since the phase of the iq only modifies the demodulation stage (and not the output stage), it is not so simple to use the iqs to do what I want (I figured out a convoluted way by demodulating a dc offset, but it's far from ideal)
--> Remark: If there is no particular reason for putting the phase on the demod rather than the modulation stage on the iqs, it could be a good idea to invert that

Moreover, in the end, we want to use the 2 channels of the ASGs to generate pulses with variable envelopes to be fed on a microwave IQ mixer, so having the 2 asgs output arbitrary waveforms in a synchronized way would be the ultimate goal.
---> Ultimately, we will need to trigger either via software or hardware the 2 asgs in a synchronized way. Again, maybe it's already possible, but I could not figure out how ?

lneuhaus, 2019-03-06:
1. The fact that the demodulator phase in the IQ module is variable was introduced (originally we had youf desired configuration where the modulation signal phase was variable), because if the IQ drives a highQ resonator, a change of the phase would require waiting for a few lifetimes for the oscillator's transient response to settle. So we need a good reason for changing this, and if we can solve the problem with the ASG, I would leave the IQ as it is.
2. We should add documentation for the ASG trigger. On that occasion, some of the options can be cleaned away.
3. We should simply adapt the synchronization mechanism of the IQ module to the ASG. I can do that by this weekend. So we can have a button/function that sets the phases of all asgs (and iqs if desired) to zero/start_phase. This should solve the problem, right?
4. You can probably also solve this by triggering both asgs on ext_positive_edge and setting the trigger DIO pin from 0 to 1. I think that should be possible with the HK module, but not 100 percent sure.

lneuhaus, 2019-03-12:
I have looked into this. The problem is that ASG is currently not a DSP module. Making it one is a major change which I would prefer to postpone until I have time to fix the issues that will come along with it. For the meantime, I added an example notebook to develop-0.9.3 under pyrpl/docs/example-notebooks/asg-synchronization-example.ipynb which demonstrates how one can synchronize two asg channels in the current fpga design.

SamuelDeleglise, 2019-03-14:
Great ! we tried your example... Also I didn't know but arbitrary waveforms
and burst modes seem to work perfectly !!!

closed 2020-04-04 by lneuhaus

diferhe, 2021-03-06:
I don't know if this is still relevant but, the asgs can be synchronized (even with software trigger) by resetting them both at the same time by writing directly to the register:
```python
#read the register
regval = p.rp.asg0._read(0)
#write reset command for both channels
p.rp.asg0._write(0, regval | 0x400040)
#clear reset
p.rp.asg0._write(0, regval)
```