pyrpl-fpga/pyrpl #374: Direct cable connection failed

MadSciSoCool, 2019-03-20:
I set up a direct wired connection between the redpitaya board and the computer,  I can visit redpitaya homepage with the IP address assigned, and can build a ssh connection in a terminal or using PuTTY etc. However, if I use the same IP for pyrpl GUI setting, the connection fails and paramiko raises an error suggesting ssh connection can not be established. (Windows 10 operating system, precompiled executable PyRPL GUI)
(more detailed exception log will be updated later)

treeTUkl, 2019-03-26:
Hi,
 my redpitaya board and my win7 pc are connected in the local network but i get an simillar error so i will post it in here, too. If it doesn't correlate please move my post!
I can connect to the redpitaya via Browser http://rp-xxxxx.local/ and via Ip Address, in terminal and PuTTY.
But the windows executable and the enviroment, via anaconda trough JetBrains, cant get the gui to work. 
I get the EOF error, shown below. I tryed to get more Infos via logging paramiko "pyrpl.sshshell.paramiko.util.log_to_file("paramikologsit.log") ", but i still have no clue how to solve this...
Help is Welcome !

Here are the loggs:
```

INFO:pyrpl:All your PyRPL settings will be saved to the config file
    C:\Users\TBreit\pyrpl_user_dir\config\deletme2.yml
If you would like to restart PyRPL with these settings, type "pyrpl.exe deletme2" in a windows terminal or 
    from pyrpl import Pyrpl
    p = Pyrpl('deletme2')
in a python terminal.
DEBUG:paramiko.transport:starting thread (client mode): 0x988de10
DEBUG:paramiko.transport:Local version/idstring: SSH-2.0-paramiko_2.4.2
DEBUG:paramiko.transport:Remote version/idstring: SSH-2.0-OpenSSH_7.2p2 Ubuntu-4ubuntu2.2
INFO:paramiko.transport:Connected (version 2.0, client OpenSSH_7.2p2)
DEBUG:paramiko.transport:kex algos:['curve25519-sha256@libssh.org', 'ecdh-sha2-nistp256', 'ecdh-sha2-nistp384', 'ecdh-sha2-nistp521', 'diffie-hellman-group-exchange-sha256', 'diffie-hellman-group14-sha1'] server key:['ssh-rsa', 'rsa-sha2-512', 'rsa-sha2-256', 'ecdsa-sha2-nistp256', 'ssh-ed25519'] client encrypt:['chacha20-poly1305@openssh.com', 'aes128-ctr', 'aes192-ctr', 'aes256-ctr', 'aes128-gcm@openssh.com', 'aes256-gcm@openssh.com'] server encrypt:['chacha20-poly1305@openssh.com', 'aes128-ctr', 'aes192-ctr', 'aes256-ctr', 'aes128-gcm@openssh.com', 'aes256-gcm@openssh.com'] client mac:['umac-64-etm@openssh.com', 'umac-128-etm@openssh.com', 'hmac-sha2-256-etm@openssh.com', 'hmac-sha2-512-etm@openssh.com', 'hmac-sha1-etm@openssh.com', 'umac-64@openssh.com', 'umac-128@openssh.com', 'hmac-sha2-256', 'hmac-sha2-512', 'hmac-sha1'] server mac:['umac-64-etm@openssh.com', 'umac-128-etm@openssh.com', 'hmac-sha2-256-etm@openssh.com', 'hmac-sha2-512-etm@openssh.com', 'hmac-sha1-etm@openssh.com', 'umac-64@openssh.com', 'umac-128@openssh.com', 'hmac-sha2-256', 'hmac-sha2-512', 'hmac-sha1'] client compress:['none', 'zlib@openssh.com'] server compress:['none', 'zlib@openssh.com'] client lang:[''] server lang:[''] kex follows?False
DEBUG:paramiko.transport:Kex agreed: ecdh-sha2-nistp256
DEBUG:paramiko.transport:HostKey agreed: ssh-ed25519
DEBUG:paramiko.transport:Cipher agreed: aes128-ctr
DEBUG:paramiko.transport:MAC agreed: hmac-sha2-256
DEBUG:paramiko.transport:Compression agreed: none
DEBUG:paramiko.transport:starting thread (client mode): 0xaa81b00
DEBUG:paramiko.transport:Local version/idstring: SSH-2.0-paramiko_2.4.2
DEBUG:paramiko.transport:Remote version/idstring: SSH-2.0-OpenSSH_7.2p2 Ubuntu-4ubuntu2.2
INFO:paramiko.transport:Connected (version 2.0, client OpenSSH_7.2p2)
DEBUG:paramiko.transport:kex algos:['curve25519-sha256@libssh.org', 'ecdh-sha2-nistp256', 'ecdh-sha2-nistp384', 'ecdh-sha2-nistp521', 'diffie-hellman-group-exchange-sha256', 'diffie-hellman-group14-sha1'] server key:['ssh-rsa', 'rsa-sha2-512', 'rsa-sha2-256', 'ecdsa-sha2-nistp256', 'ssh-ed25519'] client encrypt:['chacha20-poly1305@openssh.com', 'aes128-ctr', 'aes192-ctr', 'aes256-ctr', 'aes128-gcm@openssh.com', 'aes256-gcm@openssh.com'] server encrypt:['chacha20-poly1305@openssh.com', 'aes128-ctr', 'aes192-ctr', 'aes256-ctr', 'aes128-gcm@openssh.com', 'aes256-gcm@openssh.com'] client mac:['umac-64-etm@openssh.com', 'umac-128-etm@openssh.com', 'hmac-sha2-256-etm@openssh.com', 'hmac-sha2-512-etm@openssh.com', 'hmac-sha1-etm@openssh.com', 'umac-64@openssh.com', 'umac-128@openssh.com', 'hmac-sha2-256', 'hmac-sha2-512', 'hmac-sha1'] server mac:['umac-64-etm@openssh.com', 'umac-128-etm@openssh.com', 'hmac-sha2-256-etm@openssh.com', 'hmac-sha2-512-etm@openssh.com', 'hmac-sha1-etm@openssh.com', 'umac-64@openssh.com', 'umac-128@openssh.com', 'hmac-sha2-256', 'hmac-sha2-512', 'hmac-sha1'] client compress:['none', 'zlib@openssh.com'] server compress:['none', 'zlib@openssh.com'] client lang:[''] server lang:[''] kex follows?False
DEBUG:paramiko.transport:Kex agreed: ecdh-sha2-nistp256
DEBUG:paramiko.transport:HostKey agreed: ssh-ed25519
DEBUG:paramiko.transport:Cipher agreed: aes128-ctr
DEBUG:paramiko.transport:MAC agreed: hmac-sha2-256
DEBUG:paramiko.transport:Compression agreed: none
DEBUG:paramiko.transport:kex engine KexNistp256 specified hash_algo <built-in function openssl_sha256>
DEBUG:paramiko.transport:Switch to new keys ...
DEBUG:paramiko.transport:Adding ssh-ed25519 host key for rp-XXXXXX.local: b'XXXXXXXXXXXXXXXXXXXXXXX'
DEBUG:paramiko.transport:kex engine KexNistp256 specified hash_algo <built-in function openssl_sha256>
DEBUG:paramiko.transport:Switch to new keys ...
DEBUG:paramiko.transport:userauth is OK
INFO:paramiko.transport:Authentication (password) successful!
DEBUG:paramiko.transport:[chan 0] Max packet in: 32768 bytes
DEBUG:paramiko.transport:Received global request "hostkeys-00@openssh.com"
DEBUG:paramiko.transport:Rejecting "hostkeys-00@openssh.com" global request from server.
DEBUG:paramiko.transport:[chan 0] Max packet out: 32768 bytes
DEBUG:paramiko.transport:Secsh channel 0 opened.
DEBUG:paramiko.transport:[chan 0] Sesch channel 0 request ok
DEBUG:paramiko.transport:[chan 0] Sesch channel 0 request ok
[... cut ...]
  File "C:\ProgramData\Anaconda3\lib\site-packages\paramiko\transport.py", line 1790, in _send_user_message
    self._send_message(data)
  File "C:\ProgramData\Anaconda3\lib\site-packages\paramiko\transport.py", line 1766, in _send_message
    self.packetizer.send_message(data)
  File "C:\ProgramData\Anaconda3\lib\site-packages\paramiko\packet.py", line 413, in send_message
    self.write_all(out)
  File "C:\ProgramData\Anaconda3\lib\site-packages\paramiko\packet.py", line 354, in write_all
    raise EOFError()
EOFError

Process finished with exit code -1

```

treeTUkl, 2019-03-26:
i found an simmilar issue on git hub:https://github.com/ContinuumIO/anaconda-issues/issues/10592
and could solve my issue by installing OpenSSL(https://slproweb.com/products/Win32OpenSSL.html) with the Option Copy openSSL DLLs to: Windows system directory.

MadSciSoCool, 2019-03-30:
@treeTUkl Thank you for your comments. However, I think our issues are different. Today I set up LAN and tested again. The error log I get this time is much clearer and goes as:
```python
ERROR:paramiko.transport:Unknown exception: DLL load failed: The specified module could not be found.
ERROR:paramiko.transport:Traceback (most recent call last):
ERROR:paramiko.transport:  File "site-packages\paramiko\transport.py", line 1890, in run
ERROR:paramiko.transport:  File "site-packages\paramiko\transport.py", line 1988, in _negotiate_keys
ERROR:paramiko.transport:  File "site-packages\paramiko\kex_ecdh_nist.py", line 32, in start_kex
ERROR:paramiko.transport:  File "site-packages\paramiko\kex_ecdh_nist.py", line 51, in _generate_key_pair
ERROR:paramiko.transport:  File "site-packages\cryptography\hazmat\backends\__init__.py", line 15, in default_backend
ERROR:paramiko.transport:  File "c:\projects\pyinstaller\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
ERROR:paramiko.transport:  File "site-packages\cryptography\hazmat\backends\openssl\__init__.py", line 7, in <module>
ERROR:paramiko.transport:  File "c:\projects\pyinstaller\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
ERROR:paramiko.transport:  File "site-packages\cryptography\hazmat\backends\openssl\backend.py", line 53, in <module>
ERROR:paramiko.transport:  File "c:\projects\pyinstaller\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
ERROR:paramiko.transport:  File "site-packages\cryptography\hazmat\bindings\openssl\binding.py", line 13, in <module>
ERROR:paramiko.transport:ImportError: DLL load failed: The specified module could not be found.
ERROR:paramiko.transport:
```
I've got many hits on this error and trying to figure it out now.

lneuhaus, 2019-03-30:
Thanks for figuring out the solution! It seems that this should be directly
fixed in paramiko. Is anyone using an older Redpitaya os version? I would
guess that this problem is not present with older Redpitaya os versions on
the SD card...

MadSciSoCool, 2019-04-03:
I cloned the source code and set up the environment, then no errors occur, both in LAN and in direct connection case. I guess the original issue is due to some package mismatches in the precompiled version.

closed 2019-04-03 by MadSciSoCool