pyrpl-fpga/pyrpl #375: pdh locking using lockbox

MadSciSoCool, 2019-04-20:
Hi,
I want to use the lockbox module to do a PDH locking. I would like to ask some simple questions about the lockbox software widget since I haven't found much documentation yet.
1) What status does the is_locked's color indicate? Right now I have seen green, red, yellow (maybe also orange).
2) How does the software decide the point to be locked on? 
I did a pdh locking (lockbox set to FP cavity, using pdh as input, half ramp sweeping), and got a pdh signal at the lockbox input. I tried to lock the pdh signal to 0, but I think the laser is locked to the sideband or not locked at all, because the reflection photo-detector signal is not dropped but the error signal goes to zero and can remain with small perturbation. 
So how does the program decide where to lock, since there are multiple zeros on a pdh curve?
(I also tried to adjust the sweeping signal, leaving only the central negative slope of a pdh signal in a single sweeping. But it seems the laser is not locked to the desired peak as well.)
Thank you very much!

MadSciSoCool, 2019-04-25:
Update:
I have performed some successful lockings yesterday. It turns out to be some wrong PID coefficients. Also, I have figured out the first question. Sorry that I ignored the status bar.
However, I’m still curious about the second question. Does the program fit the signal with expected function?
P.S. Thank you for your amazing software and hardware design, saving us lots of efforts.

sixlittleone, 2022-05-10:
> Hi, I want to use the lockbox module to do a PDH locking. I would like to ask some simple questions about the lockbox software widget since I haven't found much documentation yet.
>
> 1. What status [...]

您好，您的问题解决了吗？

brobertson-ORCA, 2023-06-23:
Hi,

I am also having trouble getting a PDH lock to work. I can see the carrier and sidebands on the scope widget, but I don't understand how to get the lockbox widget to work. I've tried with both the PDHinterferomter and the FabryPerot presets. I would expect at some point to have an error signal with three zero-crossings corresponding to the derivative of the carrier and two sidebands. Has anyone successfully done a PDH lock and do you have any pointers? I have gone though the [documentation](https://pyrpl.readthedocs.io/en/latest/gui.html) and the [YouTube video](https://www.youtube.com/watch?v=WnFkz1adhgs).

Thanks.