pyrpl-fpga/pyrpl #379: nosetests failure (Error occured in reading attempt 0...)

SilasRo, 2019-06-05:
Hello, 
I have tried to install PyRPL on a Computer. First everthing looks fine (the computer successfully connects to the Red Pitaya and at the pyprl-oszi I get an input signal, but when I am trying to switch on the output of the RedPitaya (e.g. a  simple sine signal) it does not work. Instead I am getting Error messages of the following kind: 

ERROR:pyrpl.redpitaya_client:Error occured in reading attempt 0. Reconnecting at addr 0x40210000 to _writes value [    0     3     6 ... 16375 16378 16381] by client 12

It does not matter whether I start pyrpl from the "big-executable-file" (pyrpl-linux) or from a python script. Also I already have tried to install pyprl on a different Linux18 Computer. There it works completely fine. (But I can not use this computer in the laboratory...)

Since I have no idea any more, I made a nosetest, where this kind off Error messages appear, too. I would appreciate it, If someone could take a quick look at it. And maybe someone has an idea to help me. 
   
[nosetests.txt](https://github.com/lneuhaus/pyrpl/files/3256211/nosetests.txt)
If you need more information I will try to get them for you. 

Kind Regards
Silas

closed 2019-06-06 by SilasRo

SilasRo, 2019-07-26:
See issue #343 for more information.

lneuhaus, 2019-07-26:
This commit also fixes the bug: https://github.com/lneuhaus/pyrpl/commit/34a3c12284d64dc343d48f909110556391f312c3