pyrpl-fpga/pyrpl #380: using pyrpl.py _clear()

treeTUkl, 2019-06-25:
Hi
i'm seraching for an method to close pyrpl quiet and nicely.
So im trying to close pyrpl by calling the _clear() method but then i'm ending up getting an **OSError: Socket is closed.** from paramiko. The Gui freezes but is still up.

Any ideas or solutions how to close pyrpl correctly via python? 

My Code:
```python3
from pyrpl import pyrpl
if __name__ == "__main__":
    p = pyrpl.Pyrpl(config='test19_05_03')
    print("p is running! ")
    p._clear()
```
Debugger Console:
```
"C:\Program Files\Anaconda\envs\pyrpl-env-3.6\python.exe" "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.5\helpers\pydev\pydevd.py" --cmd-line --multiproc --save-signatures --qt-support=pyqt5 --client 127.0.0.1 --port 52431 --file C:/Users/(...)/pyrpl_user_dir/start.py -gui=qt
pydev debugger: process 19680 is connecting
Connected to pydev debugger (build 183.5912.18)
INFO:pyrpl:All your PyRPL settings will be saved to the config file
    C:\Users\(...)\pyrpl_user_dir\config\test19_05_03.yml
If you would like to restart PyRPL with these settings, type "pyrpl.exe test19_05_03" in a windows terminal or 
    from pyrpl import Pyrpl
    p = Pyrpl('test19_05_03')
in a python terminal.
C:\Program Files\Anaconda\envs\pyrpl-env-3.6\lib\site-packages\paramiko\rsakey.py:129: CryptographyDeprecationWarning: signer and verifier have been deprecated. Please use sign and verify instead.
  algorithm=hashes.SHA1(),
INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname xxx.xxx.xxx.xx.
p is running! 
ERROR:pyrpl.redpitaya_client:Error occured in reading attempt 0. Reconnecting at addr 0x40100014 to _reads value 1 by client 1
ERROR:pyrpl.redpitaya:Server not responding...
Traceback (most recent call last):
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya_client.py", line 154, in try_n_times
    value = function(addr, value)
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya_client.py", line 116, in _reads
    data += self.socket.recv(length * 4 - len(data) + 8)
ConnectionAbortedError: [WinError 10053] Eine bestehende Verbindung wurde softwaregesteuert
durch den Hostcomputer abgebrochen
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya.py", line 403, in endserver
    self.ssh.ask('\x03') #exit running server application
  File "C:\Users\(...)pyrpl\pyrpl\sshshell.py", line 89, in ask
    return self.askraw(question + '\n')
  File "C:\Users\(...)pyrpl\pyrpl\sshshell.py", line 84, in askraw
    self.write(question)
  File "C:\Users\(...)pyrpl\pyrpl\sshshell.py", line 63, in write
    return self.channel.send(text)
  File "C:\Program Files\Anaconda\envs\pyrpl-env-3.6\lib\site-packages\paramiko\channel.py", line 767, in send
    return self._send(s, m)
  File "C:\Program Files\Anaconda\envs\pyrpl-env-3.6\lib\site-packages\paramiko\channel.py", line 1134, in _send
    raise socket.error('Socket is closed')
OSError: Socket is closed
Traceback (most recent call last):
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya_client.py", line 154, in try_n_times
    value = function(addr, value)
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya_client.py", line 116, in _reads
    data += self.socket.recv(length * 4 - len(data) + 8)
ConnectionAbortedError: [WinError 10053] Eine bestehende Verbindung wurde softwaregesteuert
durch den Hostcomputer abgebrochen
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "C:\Users\(...)pyrpl\pyrpl\hardware_modules\scope.py", line 103, in _get_rolling_curve
    if not self._module._is_rolling_mode_active():
  File "C:\Users\(...)pyrpl\pyrpl\hardware_modules\scope.py", line 557, in _is_rolling_mode_active
    return self.rolling_mode and self._rolling_mode_allowed()
  File "C:\Users\(...)pyrpl\pyrpl\hardware_modules\scope.py", line 551, in _rolling_mode_allowed
    return self.duration > 0.1
  File "C:\Users\(...)pyrpl\pyrpl\attributes.py", line 148, in __get__
    return self.get_value(instance)
  File "C:\Users\(...)pyrpl\pyrpl\hardware_modules\scope.py", line 39, in get_value
    return obj.sampling_time * float(obj.data_length)
  File "C:\Users\(...)pyrpl\pyrpl\attributes.py", line 148, in __get__
    return self.get_value(instance)
  File "C:\Users\(...)pyrpl\pyrpl\hardware_modules\scope.py", line 63, in get_value
    return 8e-9 * float(obj.decimation)
  File "C:\Users\(...)pyrpl\pyrpl\attributes.py", line 148, in __get__
    return self.get_value(instance)
  File "C:\Users\(...)pyrpl\pyrpl\attributes.py", line 1285, in get_value
    raw_value = BaseRegister.get_value(self, obj)
  File "C:\Users\(...)pyrpl\pyrpl\attributes.py", line 218, in get_value
    return self.to_python(obj, obj._read(self.address))
  File "C:\Users\(...)pyrpl\pyrpl\modules.py", line 732, in _read
    return int(self._reads(addr, 1)[0])
  File "C:\Users\(...)pyrpl\pyrpl\modules.py", line 726, in _reads
    return self._client.reads(self._addr_base + addr, length)
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya_client.py", line 97, in reads
    return self.try_n_times(self._reads, addr, length)
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya_client.py", line 165, in try_n_times
    self.restart()
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya_client.py", line 172, in restart
    port = self._restartserver()
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya.py", line 447, in restartserver
    return self.startserver()
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya.py", line 385, in startserver
    self.endserver()
  File "C:\Users\(...)pyrpl\pyrpl\redpitaya.py", line 406, in endserver
    if 'pitaya' in self.ssh.ask():
  File "C:\Users\(...)pyrpl\pyrpl\sshshell.py", line 89, in ask
    return self.askraw(question + '\n')
  File "C:\Users\(...)pyrpl\pyrpl\sshshell.py", line 84, in askraw
    self.write(question)
  File "C:\Users\(...)pyrpl\pyrpl\sshshell.py", line 63, in write
    return self.channel.send(text)
  File "C:\Program Files\Anaconda\envs\pyrpl-env-3.6\lib\site-packages\paramiko\channel.py", line 767, in send
    return self._send(s, m)
  File "C:\Program Files\Anaconda\envs\pyrpl-env-3.6\lib\site-packages\paramiko\channel.py", line 1134, in _send
    raise socket.error('Socket is closed')
OSError: Socket is closed
```

lneuhaus, 2019-06-25:
hi. which version are you using?

treeTUkl, 2019-06-26:
> 
> 
> hi. which version are you using?

__version_info__ = (0, 9, 3, 6)