pyrpl-fpga/pyrpl #384: Adapting code and python setup.py develop command not working

vivekh99, 2019-07-23:
1. after downloading the github to my computer, i tried to run _python setup.py develop_ command and got an error that it cannot detect the cx_freeze library. I was hoping someone could help with that since i could not find the cx_freeze library in the download (although i saw it online)

2. I wanted to adapt the code to control a closed-loop controller for a mechanical oscillator (e.g. MEMS gyroscope) that has an oscillation frequency of ~10 kHz.  I found an example of a control loop at the following website: 

https://www.zhinst.com/sites/default/files/zi_appnote_mems_gyroscope.pdf

I want to build a control loop with the following features: 1) connect the voltage output from a mechanical oscillator to the voltage input port of Red Pitaya 2) let VCO to lock on the phase of the mechanical oscillator, 3) perform IQ demodulation to measure the oscillation amplitude of the mechanical oscillator, 4) connect the output of a demodulator to a PID controller to generate a control signal, and 5) create out signal by modulating the output signal from the PID controller with the VCO that is locking to the the phase of the mechanical oscillator.

lneuhaus, 2019-07-24:
sounds all very nice! A few remarks that might help you gain speed:

1) Please use the develop_0.9.3 branch of pyrpl. That one is simply more recent than the master branch and thus has a number of bugfixes for issues that you should avoid to stumble over. I simply have no time to properly merge that branch into master right now.

2) Your application idea has been on my to-do list for a long time. If you manage to implement something in this direction (even e.g. a jupyter notebook with configuration examples), do not hestiate to make a pull-request to this pyrpl repository.

3) The currently missing element in Pyrpl is a VCO. From your post, it is unclear whether you want to use a digital VCO or you have an analog one. If you have an analog one, pyrpl is totally able to solve your problem, and this has been done e.g. in https://tel.archives-ouvertes.fr/tel-01467924v1, Fig. 5.23. It is also not difficult to implement a VCO in pyrpl, and people have done this before (requires FPGA changes). An easy way to do this is to take an input signal, route this into an IQ module and add that signal (with a scale factor) to the register defining the frequency of the modulation/demodulation frequency. Let me know if you want to do this and encounter issues.

vivekh99, 2019-07-24:
Thank you for your response!

We wanted to implement a digital VCO, and are looking to download Vivado to do this, as suggested.
1) Does the Vivado version matter, must it be version 2015.4? Also, do we need to make any additional downloads other than the default packages to make a digital VCO change?

2)  Can you suggest what kind of "FPGA changes" we would need to make if we wanted to use a digital VCO.  The ultimate goal is to make a digital PLL that locks to an input signal at 10 kHz.

3) If you have any example of a digital VCO for a DPLL with an FPGA for Pyrpl, I'd appreciate it if you could share it with me.

 Thank you again I really appreciate your help.

lneuhaus, 2019-07-25:
1: version must be 2015.4
2: see above, modify iq module sine generator by adding input to frequency register
3: dont have one, sorry

vivekh99, 2019-07-31:
Hello again,

For the past week I've been trying to look over the /fpga/rtl/ code in hopes of trying to make the fpga code change that you have suggested, however I keep running into roadblocks on the meanings behind some verilog files. I am a little inexperienced in making changes at this level and am a little unsure in what file to start editing the code.

You said to route an input signal into the IQ module, so I took a look at the red_pitaya_iq_block.v file and was having trouble locating the register defining the frequency of the modulation/demodulation frequency. Also I am unsure on how exactly to re-route the input signal. I was trying to look at the block diagrams in Vivado but was not able to understand as the block diagrams didn't show an IQ block or anything resembling that.

Thank you

lneuhaus, 2019-07-31:
- https://github.com/lneuhaus/pyrpl/blob/4cb6903b7706b059d615b4fba6168fd006da2b7a/pyrpl/fpga/rtl/red_pitaya_iq_block.v#L204 - this line is the register with the phase shift per FPGA clock cycle (=sine wave frequency) to which you should add a scaled input register
- sorry the code is so poorly documented. you are definitely ambitious to touch it. Lack of time prevents me from documenting it better. If you learn something, do not hesitate to submit a PR with verilog code documentation. 
- ask yourself where the signal to control the oscillator frequency should come from. make an extra input for the red_pitaya_iq_block module for a 14-bit wide signed signal (your shift input). in red_pitaya_dsp_block.v, connect the output_signal of the PID module that you would like to use to control the VCO to the new input of an iq_block (or to the new input of all iq_blocks for simplicity). 
- ask yourself how much frequency shift a +1 V equivalent PID signal should produce. Rescale the new iq_block input signal accordingly using a bit shift, and replace the line in above linke by one where the rescaled signal is added.

vivekh99, 2019-08-07:
Hello again,

So I spent some time implementing the FPGA changes you had suggested in _red_pitaya_iq_block.v_ and _red_pitaya_dsp.v_ by adding a 14 bit **shift_input [14-1:0]** to the _red_pitaya_iq_block.v_ (at line 70) and connecting that new input to the output_signal of the pid module (that was instantiated in the dsp.v file). The output of the PID module was connected in the dsp.v file using the syntax  **.shift_input  ( output_signal[j] )** inside the _red_pitaya_iq_block_ at line 392 of the red_pitaya_dsp.v file. Lastly, I connected the shift_input to the shift_phase in _red_pitaya_iq_block.v_. 
(I attached the implementations below)

I compiled the code (using make.bat in the windows terminal) and it was able to complete the entire compilation without an ERROR and I was able to open the GUI from Spyder. 

My question now is how to test this code from the front end (or should I make front-end changes). I'm not exactly sure how you have programmed the interaction between the front-end python code and the verilog code (getting and setting pins/ports/values), and was wondering where I should begin to look.

If I am able to successfully implement this, I definitely will make a PR! 
Thank you for your help.

[image]
[image]

SamuelDeleglise, 2019-08-08:
Hi Vivek, 

Congratulations, I think you are on the right track... From what I understand, the connections between the PID and your new IQ shift register are hard-coded, such that if everything is wired properly, outputting something on the right PID output would directly shift the IQ frequency in a transparent manner for the python code.

However, I believe you didn't connect the right signal into your shift_register at line 416 of redpitaya_dsp.v

First of all, if this is going to be a hard-coded connection, it should rather be a single connection that goes from ONE of the pids to ONE of the iqs. I believe you should not make this connection in the loop of line 390. Your chance here is that the third iq (iq2) is already a customized one that is defined separately from the 2 others (see the loop with only 1 iteration at line 423). The reason for this is to have the 2 quadrature outputs taken care of by the DSP multiplexer for this particular iq module (see the extra line .signal2_o line). Since this third IQ is already an "exception", you could also add there your hard-coded connection with ONE GIVEN pid output: I would add the following line:

```
.shift_input( output_signal[3]) # if you look at the loop defined at line 320, j=2 should correspond to the third pid module pid2.
```

I believe with that, you can directly use the GUI to send some signal through PID2 (set a p-gain of 1 and use an asg as the input of pid2. send something reasonable with the ASG, then choose the right frequency that you want to modulate on your iq2, and see if the output ends up being modulated) all this should be debugged step-by step by monitoring the various signals with the scope.

Let us know how it goes... 

As I final remark: I guess the cleaner solution for the long term would be either:
- to have this modulation input taken care of by the DSP multiplexer such that you can connect an arbitrary signal to this input_shift. However, I believe having 2 extra inputs per iq modules is impossible given the current size of the multiplexer.
-  Having a switch inside the iq_module that decides wether the input signal is to be demodulated or if it is to be used as a freq_modulation input.

vivekh99, 2019-08-12:
Hello again,

So I made the suggested code changes to the program and it compiled without any errors(**i posted the updated code below just for clarity**). I had a couple questions on a few things you had mentioned as well as on the testing.

1) Can you explain why for the output_signal[_3_] you chose _3_ as your pid module, since if this was starting the index at 0, wouldn't _2_ reference the 3rd pid module. 

2) I updated the bit file (by using output_signal[_2_] pid module).  So far I can't see phase modulation from iq2 block when I examine the signal using the software scope.  I also found that after I updated the bit file, demodulation of a signal generated by the arbitrary signal generator with a local oscillator using iq0 does not work either.  I will keep on debugging the system.  I'd greatly appreciate if you can provide me any suggestion and guidance.

3) (After I the VCO issue is eventually fixed) I want to bring the frequency of the local oscillator close enough to frequency of the input signal before before closing the PLL to lock the phase and frequency of the internal oscillator to the frequency and phase of the input signal.  The frequency resolution of the local oscillator is fairly poor (~1Hz).  Is it possible for me to increase the resolution of the local oscillator to at least  10 mHz?  I looked through all the  verilog codes, and I believe this can be potentially done by  increasing the size of the frequency register. I'd appreciate if you can teach me how to increase the frequency resolution. (The frequencies of all of my my input signal is less than 100 kHz).

Thank you very much. 

 
[image]
[image]

lneuhaus, 2019-08-12:
1. You are correct, you should put a 2 there. See this definition from `red_pitaya_dsp.v`:
```

//Module numbers
localparam PID0  = 'd0; //formerly PID11
localparam PID1  = 'd1; //formerly PID12: input2->output1
localparam PID2  = 'd2; //formerly PID21: input1->output2
localparam PID3  = 'd3; //formerly PID22
localparam TRIG0  = 'd3; //formerly PID3
```
With other words, there is no more PID3, it has been changed into an instance of the trig module.

2. I cannot help you here because I am unaware of the rest of your code. Maybe post your full project on github (your account), or at least the files you changed, e.g. the _dsp.v, iq_block.v, fgen.v and so on?

3. The resolution of the iq frequency setting is of the order of 30 mHz. Changing that is a little tricky. There is an issue where I commented this i think.

lneuhaus, 2019-08-12:
This is the issue explaining the modifications needed for 3: https://github.com/lneuhaus/pyrpl/issues/372

vivekh99, 2019-08-13:
Hello again,

1. As for the rest of the code, the pictures attached to the previous post show the only code that has been changed so far. A little unsure where to proceed from there, any hints would be appreciated. I will post the full code on my GitHub page under repo name "pyrpl_vivek". The only changes that have been made so far are in the red_pitaya_dsp.v file and red_pitaya_iq_block.v file (as shown in the pics of the previous post). To view the changes, if you press CTRL + F and type "vivek" you can see where I edited the code in both files.

2. Tried playing around with the code to implement the change in the frequency resolution, however could not get that to work, but will continue trying to debug.

Thank you!

michaelcroquette, 2026-10-01:
Closing as part of a triage of long-inactive issues: PyRPL no longer uses setup.py / cx_Freeze. It is installed from pyproject.toml with pip or uv, e.g. `pip install -e .` for development.

closed 2026-10-01 by michaelcroquette