pyrpl-fpga/pyrpl #385: Opening PyRPL in Vivado 2015.4

ener-mic, 2019-07-29:
Hello,

(this is all using branch pyrpl-developer-0.9.3)

I am currently trying to open PyRPL in Vivado 2015.4 in order to view the block diagrams and eventually make FPGA changes for what I am working on. However, I am running into some difficulty when doing this. The steps I have taken are:

1) Download Vivado 2015.4 and generate licences
2)At the start screen, open tools --> run TCL script --> select red_pitaya_vivado_project.tcl

a project is generated but no block diagrams are shown (Open block design button is grayed out)

3) I then navigated to the proper directory in the Tcl console  (.../pyrpl/fpga) and and ran red_pitaya_vivado_project.tcl (running red_pitaya_vivado.tcl appears to do the same thing) which opened up a block diagram that looked like attached(see attached picture) in another window. **I am wondering if this is the entire project, or what I need to do to view the entire project (block diagrams etc.).**

**Lastly, if I wanted to begin to edit the FPGA code, can you give advice on what is the top level module file from where I should begin.**

Thank you for your help.

[image]

lneuhaus, 2019-08-12:
Everything is explained here and the links therein: https://pyrpl.readthedocs.io/en/latest/developer_guide/fpga_compilation.html

I only use the command-line mode, never open the Vivado GUI. But I think that is possle too, just cannot support you with this.

The top-level fpga module is contained in `pyrpl/fpga/rtl/red_pitaya_top.v`.

lneuhaus, 2019-08-12:
@anyone who uses the Vivado GUI, anything to help getting that up would be appreciated here.

closed 2019-11-19 by lneuhaus