pyrpl-fpga/pyrpl #391: Adding 3rd channel to Scope module

TigraneCM, 2019-11-12:
Hello all,

I'm trying to add a third channel to the scope module of pyrpl.
The main objective is to be able to display a sweeping ramp (generated at out1 of the red pitaya), a photodiode input and its demodulated error signal all at the same time, and eventually display both the input and demodulated as a function of the sweep voltage in a kind of x-y mode.

I managed to get it into the GUI without too much trouble, and through processing, but I don't know where to open an InputSelectRegister for the data itself.
Apparently the registers for ch1 and ch2 of the scope are the dsp registers of asg0 and asg1, but then I don't know which one to use for a third channel.

Do you have any ideas ?

Thanks for the help,
Tigrane (post-doc at EPFL)

lneuhaus, 2019-11-19:
The change you desire requires significant FPGA changes, especially as the current FPGA image would not have the FPGA resources to support a third scope channel. With other words, you would have to throw out some other functionality before being able to add a third scope channel. 

Also, what you request could probably be solved easily in software. If you look at the output with the scope, and trigger on the asg, you should find that you this output does not really change, i.e. you know it without having to measure it. So I suggest you simply do a ch1 + ch2 measurement in 
- a jupyter notebook
- another custom gui
- an additional software module that you write in the pyrpl api

where you display the combination of data as you desire, after having made a "model" for what signal is generated by the asg as a function of the setting. The asg ouptut buffer is accessible through `pyrpl.rp.asg0.data`, you only need to get the time axis right (usually, with `scope.trigger_source='asg0'` and `scope.trigger_delay=0`, zero time should conincide with the first buffer index, and the length of the buffer should correspond to the period `1/asg.frequency`).

TigraneCM, 2019-11-21:
Ok thanks for your answer !
I'll try to work something out along the lines you suggest and have it in the pyrpl GUI.
Indeed, I will not put my hands inside the FPGA programming, that's kind of my limit ^^

SamuelDeleglise, 2019-11-21:
If that helps, I have already written a "math Channel" in the scope gui where you plot arbitrage fonctions of in1 ans in2...
I think it s in the branche python3_only

TigraneCM, 2019-11-22:
Yes, I think I have seen this math channel feature, but I don't think this helps me.
I just want to display both in1 and iq0 as a function of out1 ( generated by asg1).

That would allow me to display a saturated spectroscopy signal as shown here :
https://www.toptica.com/fileadmin/Editors_English/03_products/03_tunable_diode_lasers/04_control_electronics/00_electronics_produktbilder/Toptica_Electronics_DLC_pro_display.png

aisichenko, 2020-03-04:
I have a similar problem, I am trying to monitor an error signal on in1 and a photodetector signal on in2 and also plot the control signal in real time. my control signal has the PID output, but I cannot figure out how to access some sort of output data array for what the PID is outputting. I would appreciate any suggestions, thanks!