pyrpl-fpga/pyrpl #397: using the library without the GUI

gsteele13, 2020-03-20:
Hi,

Is there a way to disable all GUI functionality completely? 

I'm building a GUI application inside a Jupyter notebook for interfacing my RP. I'd like to use PyRPL for communciating with it. However, I would like to avoid the PyRPL gui altogether.

Right now, if I create a `Pyrpl()` object, it always opens up a GUI popup window, which I'd really like to avoid if possible.

I've tried `gui=False`: this does disable the popup window that asks you for the location of the config file (asking you instead inline in the notebook interactively, which I would also like to avoid...but I can do that by specifying the config file...). But it does not disable the GUI popup window that asks you to add modules. As I want to use the library only as a programming interface to the RP, I would like to avoid any GUI popup windows. 

Is there a way to do this? 

(I'll also continue to read through the code in the meantime...perhaps answering my own question here :))

Cheers,
Gary

SamuelDeleglise, 2020-03-20:
What you are describing sounds like a bug...

Can you check whether the line 302 of pyrpl.py is evaluating to False (as it should)

        if self.c.pyrpl.gui:
            self.show_gui()

Also, you might have seen that all pyrpl settings are saved in a config file to stay the same from one session to the next. The config file is located (on windows systems) in c:/users/you/pyrpl_user_dir/configs/whatever_config.yml 
The config file is human readable, you might want to check the value of the gui flag in there.

Sorry that I cannot do more debugging myself, but I don't have all the good programming environment (including a redpitaya) with me in confinement...

Let me know what you find

gsteele13, 2020-03-20:
I haven't had a chance yet to check the line of the code you mention but I did edit the config file and found that gui was false for the red pitaya section but true for the pyrpl section

I edited it to be false and now I no longer have the popup window

I do get an extra "python" process in my system tray (running on macos), which has the menu items "pyrpl". I guess this is a QT app that you guys use for the QT event loop. But no gui window so that is what I wanted! 

Cheers,
Gary

closed 2020-03-20 by gsteele13