pyrpl-fpga/pyrpl #417: numpy fftpack deprecation

Bruyant, 2020-07-30:
Spectrum analyser is not starting due to depreciation of fftpack in numpy >1.17:

"NumPy’s FFT implementation was changed from fftpack to pocketfft, resulting in faster, more accurate transforms and better handling of datasets of prime length. See below under Improvements."

source

https://numpy.org/doc/stable/release/1.17.0-notes.html?highlight=numpy%20fft%20fft

my settings
     active environment : pyrpl-env
    active env location : /opt/anaconda3/envs/pyrpl-env
    conda version : 4.8.3
    conda-build version : 3.18.11
   python version : 3.7.6.final.0
   numpy                     1.18.5           py38h1da2735_0  
   numpy-base                1.18.5           py38h3304bdc_0  

Solution:
    forcing numpy <1.17
    moving to pocketfft: personally I'll try this and maybe submit a PR if it work

Bruyant, 2020-07-30:
a proposition of fix but it's not backward compatible with old numpy...

https://github.com/Bruyant/pyrpl/commit/a4edf5899a81e45438721398fff2b5a2e23e0f03

lneuhaus, 2020-11-17:
Thanks so much! I implemented your fix, and will soon merge it into master!

commit c0ae1800 (2020-11-17): scipy.fftpack->scipy.fft thanks to @Bruyant closes #417
closed 2020-11-17 by lneuhaus