pyrpl-fpga/pyrpl #418: Start_Phase attribute of ASG is not recognized

tizme555, 2020-09-03:
Hi all,

I have been trying to follow along with the API examples shown at the Pyrpl website and have run into an issue of start_phase not being recognized as a setup attribute for the ASG. This is especially important for me as I need to maintain a consistent starting point for my project. Any recommendations for fixing this is appreciated. I've attached a snapshot of the error (ignore the next line as I disconnected the red pitaya). 

[image]

lneuhaus, 2020-09-04:
Hey, 

#419 should fix this issue. Can you update your code and try again please?

Thanks
Leo

tizme555, 2020-09-04:
Hi Leo,

Thank you for your response. This solves the warning issued but by running the example code given in the API section, I do not see any changes to the resulting graph when changing start_phase. For this portion of my project, I simply have an SMA cable connected from output1 to input 1 on the red pitaya and would like to lock on to the transmitted signal where inphase will be 1/2 the amplitude of the generated signal  (or close to it) and quadrature is 0 (or close to 0).  I would like to do this for a number of frequencies and have written a python script to do so. The good news is I've managed to lock on the precise frequency generated by output1 and the results are a DC signal and the magnitude is as expected (1/2 of the transmitted amplitude) after demodulation using the IQ module. However I and Q

 are all over the place. I suspected this is because I was not specifying the start_phase but I also am now considering how the phase of the reference signal is accounted for (in the IQ module). If you, or anyone else, has a good solution to my problem I would be ecstatic. Attached is a plot of my results to demonstrate what is going on with I and Q. 

[image]

tizme555, 2020-09-04:
Even if there is an arbitrary, but consistent phase shift, I could also make that work. But the results of I and Q are not consistent.

lneuhaus, 2020-09-09:
Hey, it is somewhat unclear to me what configuration you are using, in particular what iq module and asg module do w.r.t. each other. Could you share more information, such as 

- a yaml file with your configuration 
- the code you use for the frequency scan plot

Between the lines it seems you are expecting to synchronize an IQ module with an ASG module phase. That currently does not work. However if you would use another IQ module to generate the waveform that you demodulate with another one, the phases can be synchronized with the code in develop_0_9_3 branch.

tizme555, 2020-09-14:
Hey Leo,

You are exactly right, I was trying to synchronize the IQ module with an ASG module phase. Good to know it doesn't work. I have taken your advice and have used one IQ module to generate the waveform while demodulating with another one. Thank you very much! 

I was wondering if there as any documentation on what the differences between each IQ module is. It seems that IQ module 2 outputs an Inphase and quadrature since there is IQ2 and IQ2_2 but I was curious if there are any other differences between IQ0, IQ1 and IQ2. I can't seem to find much documentation on them through the website but maybe I'm just not looking in the right place.

lneuhaus, 2020-10-20:
> I was wondering if there as any documentation on what the differences between each IQ module is. It seems that IQ module 2 outputs an Inphase and quadrature since there is IQ2 and IQ2_2 but I was [...]

There is no other difference. The proof is where these are instantiated: https://github.com/lneuhaus/pyrpl/blob/d53b829f036bb845723697c7509c91af9216a66b/pyrpl/fpga/rtl/red_pitaya_dsp.v#L370
in line 370 you see that iq0 and iq1 are identical. In line 407 you see that iq2 simply has an additional output signal connected.

closed 2020-10-20 by lneuhaus