pyrpl-fpga/pyrpl #420: IIR filter implementation incomplete?

Delphinator, 2020-09-22:
I tried to implement a controller, which is more complex than a PID, using the IIR-module. The controller is "just" a discrete-time linear transfer function. That didn't work and during debugging I stumbled across multiple issues with the IIR module I couldn't explain.

The fpga code only uses four coefficents, whereas the difference equation `y0 = b0 * x0 + b1 * x1 + b2 * x2 + a1 * y1 + a2 * y2` of a general second order section / biquad has five. There is also [a warning in the python code](https://github.com/lneuhaus/pyrpl/blob/54c3ebf8876a77925f1f16b82ba886062f1d9079/pyrpl/hardware_modules/iir/iir.py#L386) if `b2` is non-zero. This would suggest the IIR module is currently unable to implement filters with complex zeros?

I also had weird outputs suggesting an unstable filter or some sort of transient oscillation or even saturation even given very small inputs.

I'm currently prototyping a re-implementation of the fpga-side implementation - I there didn't dig too much deeper into the second issue.

lneuhaus, 2020-10-30:
Hi, please use the branch `develop-0.9.3`, the IIR implementation is a bit more advanced there than in the master branch. As far as I rememer, one can implement complex zeros, just needs to make sure the number of poles is bigger or equal to the number of zeros (for the high-frequency response not to diverge). Actually if there are as many poles as zeros, I think the code automatically adds a first-order pole near the nyquist frequency to make the filter more stable and to simplify the computation of the coeffiicients.

lneuhaus, 2020-11-17:
I just merged above branch into master. This should fix the issue. In case it persists, please re-open this issue @Delphinator

closed 2020-11-17 by lneuhaus

Delphinator, 2020-11-26:
Sorry this took so long.

As far as I can see the issue still exists. The [warning about `b_2 != 0`](https://github.com/lneuhaus/pyrpl/blob/416b2082ab8e2a448f7e9353a8fd10fb7da251ec/pyrpl/hardware_modules/iir/iir.py#L386) is still there and the [FPGA code](https://github.com/lneuhaus/pyrpl/blob/416b2082ab8e2a448f7e9353a8fd10fb7da251ec/pyrpl/fpga/rtl/red_pitaya_iir_block.v#L188-L191) only has four coefficents instead of the required five.

As far as I can tell the FPGA code implements `y0 = b0 * x0 + b1 * x1 + a1 * y1 + a2 * y2`, which corresponds to the (z-domain) transfer function

```
              b0 + b1 * z^(-1)
G(z) = ------------------------------
       1 + a1 * z^(z-1) + a2 * z^(-2)
```

which can only have a single real zero (given real-valued coefficents). It can however have one or two real poles or a single complex conjugate pair of poles.