pyrpl-fpga/pyrpl #422: Bug in custom_lockbox_example calibrate()

BobRengelink, 2020-10-22:
The custom lockbox model has a custom calibration function. Clicking the calibrate button in the GUI returns the error: "AttributeError: 'CustomInputClass' object has no attribute get_stats_from_curve".

This error can be traced to line 39 of pyrpl/software_modules/lockbox/models/custom_lockbox_example.py
Another simple to fix error is in line 37 (sweep.acquire() returns a second object)

replacing lines 33 to 39 with the following fixes the problem

```
    def calibrate(self):
        """ This is a simplified calibration method. InputSignal.calibrate works better than this in most cases. """
        self.lockbox.sweep()
        # get a curve of the signal during the sweep
        curve,_ = self.sweep_acquire()
        # fill self.mean, min, max, rms with values from acquired curve.
        self.calibration_data.get_stats_from_curve(curve=curve)
```

I would also suggest to move custom_lockbox_example.py to the pyrpl_user_dir/lockbox directory to facilitate lockbox design for non-developers

lneuhaus, 2020-10-30:
Hi Bob, thanks for the feedback. This is a good suggestion, thanks. If you'd have time to create a PR with this change that would accelerate the integration into pyrpl a lot, otherwise I'll put this on my long todo-list. Thanks!

BobRengelink, 2020-11-02:
I put up a pull request with the fix mentioned. I don't know how to edit the setup to add the code to the user_dir

lneuhaus, 2020-11-17:
merged the fix

closed 2020-11-17 by lneuhaus