pyrpl-fpga/pyrpl #423: phase lock of the ASG channels?

gsteele13, 2020-10-27:
Hi all,

Is it possible to force a lock of the phase of the two ASGs? 

I get a warning if I try to use `start_phase` in setup: 

```
WARNING:pyrpl.modules:Trying to load attribute start_phase of module asg0 that are invalid setup_attributes.
```

And furthermore, if I do this, there is an arbitrary phase offset between the channels:

```
sg_ch1.setup(frequency=1e6, amplitude=0.1, waveform='sin', trigger_source='immediately')
sg_ch2.setup(frequency=1e6, amplitude=0.1, waveform='sin', trigger_source='immediately')
```

(with or without the start_phase argument)

Is there a "sync" option to reset the phase accumulator on the two ASGs? 

Thanks,
Gary

gsteele13, 2020-10-29:
Just an update: I decided also to try this by using IQ modules to generate signals:

[image]

Unfortunately, this did not solve my problem: also with the IQ module generators, there appears to be a random, arbitrary offset between the output channels:

[image]

Does anyone know what is going on here? This is becoming now a serious limitation for us as we are aiming to do quadrature modulation using the two output channels but this is not possible if I cannot control their relative phase. 

Note that this does work using RP SCPI library ASG...we might have to switch over to that library at some point, but it would be a shame since there are so many things implemented far better here. 

Thanks,
Gary

gsteele13, 2020-10-29:
(please ignore y axis label...it should be volts...)

SamuelDeleglise, 2020-10-29:
Hi Gary, sorry for not replying sooner, I was busy with the reporting of some scientific project ;-)

I think your second approach is the correct one. There is a function to synchronize all IQs together (I believe there is even a button "synchronize IQ" in the gui, at least in the latest versions of pyrpl).

SamuelDeleglise, 2020-10-29:
I just checked, it's for instance on the branch develop_0.9.3

gsteele13, 2020-10-29:
Yep, I see it, the function call seems to be: 

```
iq.synchronize_iqs()
```

and indeed, it is not in the master branch, but is in the develop branch. I'll give it a try tomorrow.

In general, this would also be nice to have for the ASG modules, for example, if you wan to use them to generate synched pulse waveforms. (But maybe the ASGs are synced if I switch to the development branch? For tomorrow :))

Cheers,
Gary

Thanks!
Gary

lneuhaus, 2020-10-30:
https://github.com/lneuhaus/pyrpl/blob/develop-0.9.3/docs/example-notebooks/asg-synchronization-example.ipynb here is an example how this has been achieved in the past. This worked last time I tried, and the notebook will actually tell you if it doesnt work. However, I am planning on implementing this more properly in the near future.

gsteele13, 2020-10-30:
Thanks! I've checked out and installed the development branch

Using `synchronize_iqs()`, the two iq output are now phase locked:

```
iq0 = r.iq0
iq0.setup(frequency=2e6, bandwidth=[10e3,20e3], gain=0.0,
         phase=0, acbandwidth=50000, amplitude=0.5,
         input='in1', output_direct='out1',
         output_signal='quadrature', quadrature_factor=10)

iq1 = r.iq1
iq1.setup(frequency=2e6, bandwidth=[10e3,20e3], gain=0.0,
         phase=0, acbandwidth=50000, amplitude=0.5,
         input='in1', output_direct='out2', 
         output_signal='quadrature', quadrature_factor=10)

iq0.synchronize_iqs()
```

[image]

But: the `phase` argument does not seem to change the relative phase of the two outputs? If I set `phase=0` for iq0, and `phase=90`, I get exactly the same plot as above?

I'll take a  look at the asg sync notebook, thanks!

gsteele13, 2020-10-30:
I had a look too at the ASG sync code from the notebook:

```
rp.hk.expansion_P0_output = True
rp.hk.expansion_P0 = False

# setup asg trigger
for asg in [rp.asg0, rp.asg1]:
    asg.trigger_source = "ext_positive_edge"
    
# launch the trigger by creating a 0-to-1 transition on trigger pin
rp.hk.expansion_P0 = True
rp.hk.expansion_P0 = False
```

This indeed syncs my two waveforms, which is great! 

If I set the ASGs both to "sin", they are in phase, and if I set one to "sin" and the other to "cos", I get a 90 degree shift (which is what we need for our application). 

I did not  yet manage to get a phase shift working though: if I set them both to "sin" and use `phase = 90`, then I do not see a relative phase shift...?

(for us not critical right now for the application we have in mind...)

gsteele13, 2020-10-30:
(I guess  also that it should work fine to use arb waveforms and code the phase  in ourselves, I'm guessing that the phase control is based on setting the  phase register position, and then the trigger is resetting that...)

lneuhaus, 2020-11-01:
I think the variable is called start_phase, not phase. Does this change the behavior?

gsteele13, 2020-11-01:
Indeed, that is what I remember from earlier, but I think i tried it at some point because I got an error using start_phase. But indeed, start_phase seems to be a parameter of the ASGs:

[image]

In any case, also using `start_phase` doesn't solve the problem: after the trigger, the two sine waves have zero relative phase:

[image]

dermahax, 2021-03-10:
Hey, I have had the same problem. A workaround that works great for me, using the asgs, is to start one channel on the positive edge and one on the negative. However, I am working with low frequencies, I haven't tried it for faster ones.

`
phase_offset = 180

#setupt asgs
r.asg0.setup(trigger_source='ext_positive_edge')
r.asg1.setup( trigger_source='ext_negative_edge')

#launch trigger by creating a 0-1-0 transition on trigger pin
r.hk.expansion_P0 = True
time.sleep(phase_offset/360 /frequency)  #phase_offset [°]
r.hk.expansion_P0 = False`

Maybe this helps :-)