pyrpl-fpga/pyrpl #435: network analyzer docs out of date...

gsteele13, 2021-01-07:
Hi, 

Something for an overhaul at some point: the out-of-box network analyzer code from the documentation page does not work:

[image]

I'm digging in the unit tests now to find what the correct usage is, I'll post here a simple example when I work it out

gsteele13, 2021-01-07:
(I'm currently working on the python3_only branch)

gsteele13, 2021-01-07:
an update: 

I am guessing that `na.curve()` has been replaced with `na.single()`? 

This code runs: 

```
t0 = time.time()
# shortcut for na
na = p.networkanalyzer
na.iq_name = 'iq1'

# setup network analyzer with the right parameters
na.setup(start=1e3,stop=62.5e6,points=1001,rbw=1000,
         amplitude=0.2,input='iq1',output_direct='off', 
         acbandwidth=0)

#take transfer functions. first: iq1 -> iq1, second iq1->out1->(your cable)->adc1
data1 = na.single()
na.setup(input='in1', output_direct='out1')
data2 = na.single()

# get x-axis for plotting
f = na.frequencies
print("Time: %.2f" % (time.time()-t0))

plt.plot(f,np.abs(data1))
plt.plot(f,np.abs(data2))
```

But I do have a few observations:

(1) I've run this with an 8 inch SMA cable (rated to 18 GHz) from output to the input, with a tee and a 50 ohm terminator at the input, this is what I see: 

[image]

The peak around 20% of the way up seems strange? Also, the frequency axis seems very off? I think that `na.frequencies` is not working properly? 

(2) It is very slow (48 seconds, 24 seconds per trace...). If I look at an RBW of 1000 Hz (1 ms), and assume we allow for a settling time of 10 $\tau$ for the output filter, I would expect something more like 1001 ms = 1 second, but the actual acquisition seems to take 50 times longer? 

Cheers,
Gary

gsteele13, 2021-01-08:
I noticed a mistake in my calculation: 1001*1*10 = 10 seconds, so the factor is only 5...I'll dig a bit in the code to try to understand better how it works.

gsteele13, 2021-01-08:
I'm still having trouble interpreting any of the results from the NA code

I've cross checked it with the bode analyzer from the RP, and this works fine (aside from some electrical resonances probably due to the RP port impedances), but the PYRPL NA seems to give something that makes no sense?

This is a 25 MHz HP filter measured using the PYRPL NA code (same as above, with HP filter in series  

[image]

whereas this is exactly the same device measured using the Bode analyser: 

[image]

SamuelDeleglise, 2021-01-08:
Have you checked if nothing was preventing the output to work correctly ? A frequent pitfall with Pyrpl is the saturation of the output, since the output_direct from all modules are sumed before being outputed to out1 or out2. Can you check with the scope during the NA scan, that the signal of output1 looks like what you would expect ? (I would recommend using the gui for that)

Did you find where the bug with the frequency axis was coming from in the mean time ?

gsteele13, 2021-01-08:
I broke down and finally tried out the GUI (stubborn :)) 

Things look pretty good, these are my LPF and HPF in the GUI:

[image]
[image]

Will see if I can get this going in the notebook too. The GUI took me a bit of getting used to but it is a good interactive interface I must say :)

gsteele13, 2021-01-08:
the feature around 12 MHz btw seems to be an impedance issue of RP...the real HPF is very flat on my VNA:

[image]

lneuhaus, 2021-01-26:
Did this measurement use a 50-Ohm terminator both at the RP output and input?