pyrpl-fpga/pyrpl #437: NEWSLETTER

lneuhaus, 2021-01-29:
I'm putting this here since we don't have a mailing list / google group etc yet, and nobody seems to use our gitter. 

1. After more than 3 years, the branch "develop_0.9.3" got finally merged into master a few weeks ago.
2. Next on the list is dropping python2 support (https://github.com/lneuhaus/pyrpl/pull/355) - please let me know if this is a problem for you.
3. After that, the next thing I plan is to drop the IIR filter from the master branch. I won't delete the code, just deactivate the IIR filter from the default branch, since it uses so many resources that it is not compatible with other features I've implemented in the meantime (which would make their way into the master branch). There could always be a feature branch with IIR filter support, just not the master branch. Again, please let me know if this is a problem for you.
4. After that I plan a major refactor, using https://github.com/pypga/pypga as the basis for the FPGA design. Have a look at the new PyPGA library and see for yourself. My hope is that this will facilitate maintaining multiple FPGA designs in a single repository, e.g. one with and one without IIR filter. If you're willing to contribute to this migration, your help is welcome!

lneuhaus, 2021-01-29:
and here is a google group - not sure we will use it but I will give it a chance https://groups.google.com/g/pyrpl

michaelcroquette, 2026-10-01:
Closing as part of a triage of long-inactive issues: this 2021 announcement is out of date.

closed 2026-10-01 by michaelcroquette