pyrpl-fpga/pyrpl #440: Slow Analog PID

fegewers10, 2021-03-08:
Hello,

thank you so much for the amazing library. I am using one Red pitaya board with PYRPL to lock a cavity with PDH. I would like to know if it is possible to use the slow analog inputs and outputs in a PID to control the temperature. I couldn't find a way to configure the PID input for this.

Thank you
Fegewers

SamuelDeleglise, 2021-03-09:
Thanks for the feedback. This is definitely possible. I think, the trick is to configure the input of a Pwm module to be the pid that you want to use (and then setting output_direct of the said pid to off)... 

Something like (excerpt from the file https://github.com/lneuhaus/pyrpl/blob/master/pyrpl/hardware_modules/pwm.py)

    pwm0 = AuxOutput(output='pwm0') # I guess what is meant is p.rp.pwm0...
    pwm0.input = 'pid0'
    Pid(client, module='pid0').ival = 0

Let me know if this is solving your problem

fegewers10, 2021-03-09:
Thank you for the answer.
I understood that this code will associate the output of the PID0 to the output pwm0. How can I feed the input of a slow analog signal into the PID module?

Thanks again

SamuelDeleglise, 2021-03-10:
I am not sure there is currently support for slow analog inputs...

lneuhaus, 2021-03-30:
The slow analog inputs are only supported tor querying, though the signals are available in the FPGA. It would be straightforward to connect them to a PID when changing the verilog code. The reason this isn't done in the main image right now is because we have too few free "input signal" slots.

lneuhaus, 2021-03-30:
See https://github.com/lneuhaus/pyrpl/blob/ac0493f1b17abcf9a0c2322e8f5f98e6d10f4c9f/pyrpl/fpga/rtl/red_pitaya_ams.v#L355 - the slow analog inputs are available as wires here.