pyrpl-fpga/pyrpl #445: ERROR during bitfile generation

haectobi, 2021-05-05:
Hi,

When trying to do the bitfile generation of the unmodified code I get following Error:

ERROR: [DRC 23-20] Rule violation (INBB-3) Black Box Instances - Cell 'i_dsp/genblk4[4].iir/p_ay1_module/product0' of type 'i_dsp/genblk4[4].iir/p_by1_module/product0/mult_32_32' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.

I am using Vivado 2015.4 on windows 10 and saw on the Xilinx webpage that these versions dont support each other. But I am unsure because the synthesis and simulation work. Could the version be the issue?

Or has anyone else had this Error?

lneuhaus, 2021-05-05:
I never tested vivado on win10. Could certainly be the issue. Can you test on a different OS, eg Ubuntu booted from a usb stick?

haectobi, 2021-05-05:
Can you also compile pyrpl with a different Vivado version?

lneuhaus, 2021-05-05:
not without errors I think. In principle it should be straightforward to migrate to a more recent vivado version, but this will require some modifications.

closed 2021-05-06 by haectobi