pyrpl-fpga/pyrpl #463: How to deal with the strong shock

Chenelusive, 2021-12-01:
Hello 
In some experiments of locking cavity, some shocks or disturbances carry force sufficient to move the mirrors so that the laser frequency shoots beyond the error signal wings generated by IQ module.I would like to know if it is possible to generate a bias voltage to bring it back to the error signal wings before the PID module lose the control of the system,and which module can generate bias voltage?

waterwin, 2021-12-07:
The asgs has a setting for a bias called offset, maybe that can be used with smart setting of the rest of the parameters

[image]

michaelcroquette, 2026-10-22:
The idea here is to use a two stage lock where you first lock on side of the fringe before switching to the pdh signal as an error signal.

closed 2026-10-22 by michaelcroquette