pyrpl-fpga/pyrpl #467: AttributeError: module 'qtpy.QtCore' has no attribute 'pyqtBoundSignal'

ricardocabrita, 2022-02-02:
Hello,

When running the PyRPL GUI, when I try to load any module I get the error below, and nothing else happens: 
AttributeError: module 'qtpy.QtCore' has no attribute 'pyqtBoundSignal'.

I installed PyRPL via the conda command on the documentation - I'm running it on MacOS Catalina. I checked on python command line and I have PyQT5, and it has the class PyQt5.QtCore.pyqtBoundSignal, however qtpy.QtCore.pyqtBoundSignal does not exist. I'm saying this in case it helps, I thought pyqt would wrap PyQT5.

ricardocabrita, 2022-02-02:
I took a look at the my installed version of pyqt (using version 1.11.2) and it seems it renames the pyqtBoundSignal object:
`from PyQt5.QtCore import pyqtBoundSignal as SignalInstance`.
So I looked at older versions of qtpy, and I installed version 1.9.0 which doesn't rename the pyqtBoundSignal, so now I can load the modules I tried, IQ and SpectrumAnalyzer.

Is it simply the case that pyRPL is meant to run with version 1.9.0 of qtpy or did I miss something along the way ?

SamuelDeleglise, 2022-02-02:
I believe this bug has been fixed in the branch [python3-only](https://github.com/lneuhaus/pyrpl/commit/c488bba88191069795b8b87ff6054ae552759254)

ricardocabrita, 2022-02-02:
Thanks for the reply, that seems to be the case. Is there one package version that I could install with pip that compiles that branch ? Or I should just install it manually from source ?

SamuelDeleglise, 2022-02-03:
Nope, I don't think so... also @lneuhaus, I think now python3-only was merged into master... Probably we should just remerge the few new commits that were made on python3only into master and delete it definitively ?

abregnsbo, 2022-08-11:
I got the `has no attribute 'pyqtBoundSignal'` error when using pyrpl from master branch with Python 3.7, but the python3-only branch seems to work fine.

michaelcroquette, 2026-10-01:
Closing as part of a triage of long-inactive issues: this was an incompatibility between old versions of qtpy and PyQt. Current PyRPL works with recent qtpy and PyQt5, PyQt6, PySide2 or PySide6. If this still affects you with the current release (0.9.9.1 / main), please comment here and we will reopen.

closed 2026-10-01 by michaelcroquette