pyrpl-fpga/pyrpl #468: Python 3.10 not supported?

parkerlreed, 2022-02-23:
Arch Linux

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 187, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.10/runpy.py", line 146, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.10/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/home/parker/pyrpl/pyrpl/__init__.py", line 59, in <module>
    global_config = MemoryTree('global_config', source='global_config')
  File "/home/parker/pyrpl/pyrpl/memory.py", line 507, in __init__
    self._savetimer.setInterval(self._loadsavedeadtime*1000)
TypeError: setInterval(self, int): argument 1 has unexpected type 'float'
```

clarencesees, 2023-02-21:
I'm facing the same problem too.

gsteele13, 2023-08-17:
OK, rolling back to 3.8 in a new env also gave me the same error about `GraphicsWindow` 😢 

Must be a specific problem with the version of QT....will do some googling...

gsteele13, 2024-01-30:
Sure, no problem. It was a while ago, but I think (?) this was the env that worked (I have 8 copies of environment attempts...)

[pyrpl_env7.txt](https://github.com/lneuhaus/pyrpl/files/14099854/pyrpl_env7.txt)

hkosovacgodart, 2024-01-31:
Thank you very much for the env copy, I indeed had to add/remove a few libraries but it definitely gave me a solid base and I managed to open pyrpl (finally!). Thank you for your help!

michaelcroquette, 2026-10-01:
Python 3.10 and later are now supported: CI tests Python 3.9 to 3.14. Closing. If this still affects you with the current release (0.9.9.1 / main), please comment here and we will reopen.

closed 2026-10-01 by michaelcroquette