pyrpl-fpga/pyrpl #469: module 'qtpy.QtCore' has no attribute 'pyqtBoundSignal'?

Chenelusive, 2022-03-03:
[image]
[image]
I was opening a GUI with an existed yml file,and have no idea what was going on

KaifengC, 2022-04-20:
I ran into the same issue and notice someone else has the same problem: https://github.com/lneuhaus/pyrpl/issues/467

My workaround is to excute `pip install qtpy==1.9.0` and it worked.

abregnsbo, 2022-08-11:
Also ran into this issue. My workaround was using the `python3-only` branch of pyrpl instead of `master`.

michaelcroquette, 2026-10-01:
Closing as a duplicate of #467 (same qtpy `pyqtBoundSignal` error).

closed 2026-10-01 by michaelcroquette