pyrpl-fpga/pyrpl #476: Lockbox pwm outputs - max_hold_no_iir branch

tt-aqt, 2022-06-13:
Has anyone gotten these outputs to work with the Lockbox module?  I am using the max_hold_no_iir branch.

Looking in output.py, it seems like it should be possible to use pwm0 and pwm1 as outputs?   i.e. pins 17 and 18 on the E2 extension connector.  Monitoring lockbox.outputs.pwm_output with the scope seems successful.  Everything seems to work and the direct output of the pid module assigned to the pwm output gets set to off.

It's just that nothing comes out of pins 17 or 18.

Amany-k, 2025-05-14:
Hello, were you eventually able to get pwm0 or pwm1 working properly?

I’m trying to use pwm1 for temperature control to lock the frequency of a laser, but I'm encountering the same issue: the pin always outputs 0.9 V, even after I change the offset value (e.g., to -1.0 or +1.0). I’ve routed the lockbox output using temperature.output_channel = 'pwm1', and set temperature.offset = -1, but there’s no voltage change — it just stays at 0.9 V.

In the PLL example tutorial, it seems pwm1 is used successfully for temperature feedback, so I’m wondering if there’s a step I might be missing or if you were able to resolve this issue?

tt-aqt, 2025-05-14:
Yes but not with that branch.  I forget which branch the second attempt was with.  It was incredibly noisy, so i abandoned the idea.

Amany-k, 2025-05-14:
Thanks for getting back to me! Yes, I read that it is noisy and that a low-pass filter is recommended so I plan to add one externally as well.
I’d really like to give it another shot so  hopefully I will hear back from one of the coordinators on which branch currently supports pwm1 properly.

michaelcroquette, 2026-09-23:
Sorry for the late reply. Pwm output should work on the main branch. Were you able to make it work ?