pyrpl-fpga/pyrpl #477: Can't find a valid hostname for red pitaya

RAILMEMS1, 2022-06-15:
[image]

I followed the quick start steps on the red pitaya website, and after I write the image file provided on the website, I'm able to open the web interface, then I think I'm ready to start installing pyrpl.

I download the pyrpl-windows.exe and write the image file(redpitaya_v0.92.img) provided on the pyrpl website. However, when I'm running the pyrpl-windows.exe, I can't find the hostname or the mac address shown in the youtube video tutorial. Also some errors also occur as shown in the image attached.

tt-aqt, 2022-06-17:
have you tried typing the ip into the Hostname box and hitting ok?

RAILMEMS1, 2022-06-17:
> 

I tried, but the window just disappeared after seconds and nothing happened then

jackschrott, 2022-08-09:
I also see this behavior when trying to run pyrpl-windows.exe. After double clicking the executable I am prompted to pick or create a configuration file, or hit 'cancel' for no file. Whichever option I choose a gui pops up with a loading bar that says scanning LAN for red pitayas. When the loading bar gets to 42% the window closes as does the terminal window which opens on initial start of the executable. This all happens too fast to try putting in the ip of my red-pitaya as @RAILMEMS1 RAILMEMS1 points out

Tomland-Don, 2023-07-25:
Did you solve this tissue? I also cannot  find my ip address and mac address, but my pitaya is connected.

Tomland-Don, 2023-07-25:
Did you solve this problem? I also can't find the ip address and mac address, but my pitaya is connected.

mcromb, 2023-10-24:
I am also having this problem. get the paramiko.transport errors, the automatic search fails to find the ip address, when i enter it manually, the exe file quickly errors and closes. 
Some people i know were able to solve this by creating a renamed copy of global_config.yml when installing and using that instead but i have had no such luck.

gotodani, 2024-06-04:
I summarize my method for solving this issue in the attached PDF.
[Pyrpl connection.pdf](https://github.com/user-attachments/files/15545791/Pyrpl.connection.pdf)

mcromb, 2024-06-21:
> I summarize my method for solving this issue in the attached PDF. The hostname need to be type manually after setting static IP address in the Red Pitaya. [Pyrpl connection.pdf](https://github.com/user-attachments/files/15545791/Pyrpl.connection.pdf)

I tried setting the RP address static as shown but still had the same issue with the errors and it force exiting after entering a hostname. I was not able to set my computer IP static as it is not allowed on a university network

peteasa, 2024-11-01:
I have a potential fix for this problem see - https://github.com/RedPitaya/pyrpl/commit/ff7c04afca9114accb41d65e24be105db1d15da5

jilingcan, 2025-04-11:
My question:
INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname 192.168.x.xxx.
Failed to execute script __main__
LOADER: OK.
LOADER: Cleaning up Python interpreter.

why:Failed to execute script __main__

closed 2025-12-11 by michaelcroquette