pyrpl-fpga/pyrpl #478: IIR filter add more noise than should be

GuanhaoHuang, 2022-08-04:
Hi!

I have recently been trying to use IIR filter to do some feedback experiment, but found out that it adds substantial digital noise to the feedback signal. 

As a comparison, when the signal is passed through the IQ bandpass filter, no added noise is observed except the ADC DAC 14 bits noise at the input output. 

But for IIR, somehow a large flat background noise (likely some digitisation noise) is added (saw on spectrum analyzer, more than 20dB than the output DAC noise), with even higher noise at the position of the poles. The noise behaves as if the registers used for the IIR filters don't have enough bits, but after checking it should have 32 bits, higher than what was used in IQ bandpass filter 24 bits. 

I wonder if this behaviour is expected, or am I missing something in the IIR filter settings?

GuanhaoHuang, 2022-08-08:
Sorry the 20dB is an exaggeration, that's compared to the result when setting output = 0, which is just the output DAC noise. it's more like 10dB. when compared to the original input ADC noise.

abregnsbo, 2022-10-13:
I noticed that on the master branch fpga timing seems not to be closed in the iir filter block, with a large -5.2 ns violation on a 8.0 ns clock cycle, see pyrpl/fpga/out/post_synth_timing_summary.rpt.

SamuelDeleglise, 2022-10-13:
Also, I should notice that I have done a lot of debugging of the iir in the branch python3-only:

  - in particular, I change the hardware implementation of the biquads from Direct form 2 to Direct form 1, see [this](https://en.wikipedia.org/wiki/Digital_biquad_filter)

- There was also a bug I believe in the software definition of the biquads coefficients, which essentially prevented to properly define a complex zero at all

In short, I advise to use the branch python3-only untill it is merged into master

abregnsbo, 2022-10-13:
@SamuelDeleglise thanks for info. Also on master there are some minor timing setup bugs in the XDC file. Are you interested in getting these pulled into python3-only ?

SamuelDeleglise, 2022-10-13:
You mean that you have fixed some bugs ?

abregnsbo, 2022-10-13:
In the XDC, yes. Though I am new to PyRPL and Red Pitaya, and not sure if the rtl & Vivado setup files are maintained at all, or just stale files.

SamuelDeleglise, 2022-10-13:
I don't know either...
Also, in what commit are the bugfix you are talking about? Are you working on a fork of this repository ?

abregnsbo, 2022-10-13:
I am working on a local branch where the fpga will contain a time-tagger, controlled by a branched version of PyRPL. Looking at python3-only it seems you have modified the rtl file there.

michaelcroquette, 2026-09-23:
Please check the branch Reduce_fpga_issues. I have updated the IIr code.

closed 2026-09-23 by michaelcroquette